JakobEngel/dso_ros #2: compilation error

hfarazi, 2016-11-14:
rosmake output:
```
[ rosmake ] rosmake starting...                                                                       
[ rosmake ] No package specified.  Building ['dso_ros']                                               
[ rosmake ] Packages requested are: ['dso_ros']                                                       
[ rosmake ] Logging to directory /home/hafez/.ros/rosmake/rosmake_output-20161114-122352              
[ rosmake ] Expanded args ['dso_ros'] to:
['dso_ros']                                                 
[rosmake-0] Starting >>> catkin [ make ]                                                              
[rosmake-0] Finished <<< catkin ROS_NOBUILD in package catkin
 No Makefile in package catkin          
[rosmake-6] Starting >>> cmake_modules [ make ]                                                       
[rosmake-7] Starting >>> genmsg [ make ]                                                              
[rosmake-0] Starting >>> cpp_common [ make ]                                                          
[rosmake-2] Starting >>> rosgraph [ make ]                                                            
[rosmake-3] Starting >>> rosclean [ make ]                                                            
[rosmake-0] Finished <<< cpp_common ROS_NOBUILD in package cpp_common
 No Makefile in package cpp_common
[rosmake-0] Starting >>> rostime [ make ]                                                             
[rosmake-7] Finished <<< genmsg ROS_NOBUILD in package genmsg
 No Makefile in package genmsg          
[rosmake-7] Starting >>> genlisp [ make ]                                                             
[rosmake-6] Finished <<< cmake_modules ROS_NOBUILD in package cmake_modules
 No Makefile in package cmake_modules
[rosmake-5] Starting >>> genpy [ make ]                                                               
[rosmake-6] Starting >>> rospack [ make ]                                                             
[rosmake-1] Starting >>> gencpp [ make ]                                                              
[rosmake-2] Finished <<< rosgraph ROS_NOBUILD in package rosgraph
 No Makefile in package rosgraph    
[rosmake-4] Starting >>> roslang [ make ]                                                             
[rosmake-2] Starting >>> rosparam [ make ]                                                            
[rosmake-3] Finished <<< rosclean ROS_NOBUILD in package rosclean
 No Makefile in package rosclean    
[rosmake-3] Starting >>> rosmaster [ make ]                                                           
[rosmake-7] Finished <<< genlisp ROS_NOBUILD in package genlisp
 No Makefile in package genlisp       
[rosmake-7] Starting >>> xmlrpcpp [ make ]                                                            
[rosmake-0] Finished <<< rostime ROS_NOBUILD in package rostime
 No Makefile in package rostime       
[rosmake-5] Finished <<< genpy ROS_NOBUILD in package genpy
 No Makefile in package genpy             
[rosmake-0] Starting >>> roscpp_traits [ make ]                                                       
[rosmake-1] Finished <<< gencpp ROS_NOBUILD in package gencpp
 No Makefile in package gencpp          
[rosmake-1] Starting >>> message_generation [ make ]                                                  
[rosmake-2] Finished <<< rosparam ROS_NOBUILD in package rosparam
 No Makefile in package rosparam    
[rosmake-3] Finished <<< rosmaster ROS_NOBUILD in package rosmaster
 No Makefile in package rosmaster 
[rosmake-4] Finished <<< roslang ROS_NOBUILD in package roslang
 No Makefile in package roslang       
[rosmake-7] Finished <<< xmlrpcpp ROS_NOBUILD in package xmlrpcpp
 No Makefile in package xmlrpcpp    
[rosmake-0] Finished <<< roscpp_traits ROS_NOBUILD in package roscpp_traits
 No Makefile in package roscpp_traits
[rosmake-0] Starting >>> roscpp_serialization [ make ]                                                
[rosmake-1] Finished <<< message_generation ROS_NOBUILD in package message_generation
 No Makefile in package message_generation
[rosmake-6] Finished <<< rospack ROS_NOBUILD in package rospack
 No Makefile in package rospack       
[rosmake-6] Starting >>> roslib [ make ]                                                              
[rosmake-6] Finished <<< roslib ROS_NOBUILD in package roslib
 No Makefile in package roslib          
[rosmake-6] Starting >>> rosunit [ make ]                                                             
[rosmake-0] Finished <<< roscpp_serialization ROS_NOBUILD in package roscpp_serialization
 No Makefile in package roscpp_serialization
[rosmake-0] Starting >>> message_runtime [ make ]                                                     
[rosmake-0] Finished <<< message_runtime ROS_NOBUILD in package message_runtime
 No Makefile in package message_runtime
[rosmake-6] Finished <<< rosunit ROS_NOBUILD in package rosunit
[... 65 lines cut ...]
  /home/hafez/Workspace/src/dso_ros/src/main.cpp:159: undefined reference to `ros::init(int&, char**, std::string const&, unsigned int)'
  /home/hafez/Workspace/src/dso_ros/src/main.cpp:201: undefined reference to `ros::NodeHandle::NodeHandle(std::string const&, std::map<std::string, std::string, std::less<std::string>, std::allocator<std::pair<std::string const, std::string> > > const&)'
  /home/hafez/Workspace/src/dso_ros/src/main.cpp:204: undefined reference to `ros::spin()'
  /home/hafez/Workspace/src/dso_ros/src/main.cpp:202: undefined reference to `ros::Subscriber::~Subscriber()'
  /home/hafez/Workspace/src/dso_ros/src/main.cpp:209: undefined reference to `ros::NodeHandle::~NodeHandle()'
  /home/hafez/Workspace/src/dso_ros/src/main.cpp:202: undefined reference to `ros::Subscriber::~Subscriber()'
  /home/hafez/Workspace/src/dso_ros/src/main.cpp:209: undefined reference to `ros::NodeHandle::~NodeHandle()'
  collect2: error: ld returned 1 exit status
  make[3]: *** [bin/dso_live] Error 1
  make[3]: Leaving directory `/home/hafez/Workspace/src/dso_ros/build'
  make[2]: *** [CMakeFiles/dso_live.dir/all] Error 2
  make[2]: Leaving directory `/home/hafez/Workspace/src/dso_ros/build'
  make[1]: *** [all] Error 2
  make[1]: Leaving directory `/home/hafez/Workspace/src/dso_ros/build'
-------------------------------------------------------------------------------}
[ rosmake ] Output from build of package dso_ros written to:
[ rosmake ]    /home/hafez/.ros/rosmake/rosmake_output-20161114-122352/dso_ros/build_output.log
[rosmake-1] Finished <<< dso_ros [FAIL] [ 11.38 seconds ]                                             
[ rosmake ] Halting due to failure in package dso_ros. 
[ rosmake ] Waiting for other threads to complete.
[ rosmake ] Results:                                                                                  
[ rosmake ] Built 34 packages with 1 failures.                                                        
[ rosmake ] Summary output to directory                                                               
[ rosmake ] /home/hafez/.ros/rosmake/rosmake_output-20161114-122352 
```

hfarazi, 2016-11-14:
To fix it, I changed manifest.xml to following:

```
<package>
  <description brief="dso_core">

     ROS Wrapper DSO.

  </description>
  <author>Jakob Engel</author>
  <license>see http://vision.in.tum.de/dso</license>
  <review status="unreviewed" notes=""/>
  <url>http://vision.in.tum.de/dso</url>
  <depend package="std_msgs"/>
  <depend package="rospy"/>
  <depend package="roscpp"/>
  <depend package="roslib"/>
  <depend package="cv_bridge"/>
  <depend package="sensor_msgs"/>
</package>
```

23pointsNorth, 2016-11-14:
I had the same problem, so I converted it to catkin. Check out the pull request or the linked repo.

JakobEngel, 2016-11-15:
Should be fixed - thanks!

closed 2016-11-15 by JakobEngel