JakobEngel/dso_ros #5: catkin_make error

oya123, 2016-11-18:
after
`export DSO_PATH=[PATH_TO_DSO]/dso` 
`catkin_make`
output is :
`Base path: /home/ouyangqiuping/DSO/DSO_WS
Source space: /home/ouyangqiuping/DSO/DSO_WS/src
Build space: /home/ouyangqiuping/DSO/DSO_WS/build
Devel space: /home/ouyangqiuping/DSO/DSO_WS/devel
Install space: /home/ouyangqiuping/DSO/DSO_WS/install
####
#### Running command: "cmake /home/ouyangqiuping/DSO/DSO_WS/src -DCATKIN_DEVEL_PREFIX=/home/ouyangqiuping/DSO/DSO_WS/devel -DCMAKE_INSTALL_PREFIX=/home/ouyangqiuping/DSO/DSO_WS/install -G Unix Makefiles" in "/home/ouyangqiuping/DSO/DSO_WS/build"
####
-- The C compiler identification is GNU 4.9.4
-- The CXX compiler identification is GNU 4.9.4
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Using CATKIN_DEVEL_PREFIX: /home/ouyangqiuping/DSO/DSO_WS/devel
-- Using CMAKE_PREFIX_PATH: /home/ouyangqiuping/qt_opensource/devel;/home/ouyangqiuping/lsd_slam_catkin/devel;/opt/ros/indigo
-- This workspace overlays: /home/ouyangqiuping/qt_opensource/devel;/home/ouyangqiuping/lsd_slam_catkin/devel;/opt/ros/indigo
-- Found PythonInterp: /home/ouyangqiuping/.venv/python2.7/bin/python (found version "2.7.6") 
-- Using PYTHON_EXECUTABLE: /home/ouyangqiuping/.venv/python2.7/bin/python
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/ouyangqiuping/DSO/DSO_WS/build/test_results
-- Looking for include file pthread.h
-- Looking for include file pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Found gtest sources under '/usr/src/gtest': gtests will be built
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.6.18
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 1 packages in topological order:
-- ~~  - dso_ros
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin package: 'dso_ros'
-- ==> add_subdirectory(dso_ros-catkin)

---- LOOKING FOR DSO at
"/home/ouyangqiuping/DSO/dso"

---- FOUND Pangolin Headers at
"/home/ouyangqiuping/Pangolin/include;/home/ouyangqiuping/Pangolin/build/src/include;/usr/include;/usr/include;/usr/include/eigen3"
---- FOUND Pangolin Libs at
"pangolin"

---- FOUND OpenCV Headers at
"/usr/local/include/opencv;/usr/local/include"
---- FOUND OpenCV Libs at
"opencv_videostab;opencv_video;opencv_superres;opencv_stitching;opencv_photo;opencv_ocl;opencv_objdetect;opencv_nonfree;opencv_ml;opencv_legacy;opencv_imgproc;opencv_highgui;opencv_gpu;opencv_flann;opencv_features2d;opencv_core;opencv_contrib;opencv_calib3d"
-- Found Eigen3: /usr/local/include/eigen3 (Required is at least version "2.91.0") 
-- Boost version: 1.54.0
-- Found the following Boost libraries:
--   system
--   thread
-- Configuring done
[... 60 lines cut ...]
     vector<_Tp, _Alloc>::
     ^
/usr/include/c++/4.9/bits/vector.tcc:167:5: note:   no known conversion for argument 1 from ‘dso::IOWrap::PangolinDSOViewer*’ to ‘const std::vector<dso::IOWrap::Output3DWrapper*>&’
In file included from /usr/include/c++/4.9/vector:64:0,
                 from /usr/include/c++/4.9/bits/random.h:34,
                 from /usr/include/c++/4.9/random:49,
                 from /usr/include/c++/4.9/bits/stl_algo.h:66,
                 from /usr/include/c++/4.9/algorithm:62,
                 from /usr/include/eigen3/Eigen/Core:160,
                 from /home/ouyangqiuping/DSO/dso/src/util/NumType.h:27,
                 from /home/ouyangqiuping/DSO/dso/src/FullSystem/FullSystem.h:28,
                 from /home/ouyangqiuping/DSO/DSO_WS/src/dso_ros-catkin/src/main.cpp:37:
/usr/include/c++/4.9/bits/stl_vector.h:448:7: note: std::vector<_Tp, _Alloc>& std::vector<_Tp, _Alloc>::operator=(std::vector<_Tp, _Alloc>&&) [with _Tp = dso::IOWrap::Output3DWrapper*; _Alloc = std::allocator<dso::IOWrap::Output3DWrapper*>]
       operator=(vector&& __x) noexcept(_Alloc_traits::_S_nothrow_move())
       ^
/usr/include/c++/4.9/bits/stl_vector.h:448:7: note:   no known conversion for argument 1 from ‘dso::IOWrap::PangolinDSOViewer*’ to ‘std::vector<dso::IOWrap::Output3DWrapper*>&&’
/usr/include/c++/4.9/bits/stl_vector.h:470:7: note: std::vector<_Tp, _Alloc>& std::vector<_Tp, _Alloc>::operator=(std::initializer_list<_Tp>) [with _Tp = dso::IOWrap::Output3DWrapper*; _Alloc = std::allocator<dso::IOWrap::Output3DWrapper*>]
       operator=(initializer_list<value_type> __l)
       ^
/usr/include/c++/4.9/bits/stl_vector.h:470:7: note:   no known conversion for argument 1 from ‘dso::IOWrap::PangolinDSOViewer*’ to ‘std::initializer_list<dso::IOWrap::Output3DWrapper*>’
make[2]: *** [dso_ros-catkin/CMakeFiles/dso_live.dir/src/main.o] error 1
make[1]: *** [dso_ros-catkin/CMakeFiles/dso_live.dir/all] error 2
make: *** [all] error 2
Invoking "make -j4 -l4" failed
`

jingedawang, 2016-11-18:
I have the same problems as you.
Using ROS Indigo, catkin workspace.
Hope Engel's solution.

mingding1, 2016-11-18:
I also have this problems in ubuntu 14.04 and ROS Indigo. What should i do to solve this problems?

JakobEngel, 2016-11-18:
fixed it (hopefully, since I cannot test it).
I forgot to update the catkin branch with to match the newest DSO commit - sorry!

closed 2016-11-18 by JakobEngel