JakobEngel/dso_ros #6: benchmarkSpecialOption

chiragmajithia, 2016-11-19:
Hello,
I changed the CMakeLists.txt  file as suggested in #4  and I tried to compile the code, I get following error.
I am attaching the CMakeLists.txt file, for the reference.. 
`/home/chirag/vision/SLAM/src/dso_ros/src/main.cpp: In function ‘void parseArgument(char*)’:
/home/chirag/vision/SLAM/src/dso_ros/src/main.cpp:84:3: error: ‘benchmarkSpecialOption’ was not declared in this scope
   benchmarkSpecialOption = option;
`

[CMakeLists.txt](https://github.com/JakobEngel/dso_ros/files/601520/CMakeLists.txt)

jingedawang, 2016-11-21:
I have the same problem with you.

JakobEngel, 2016-11-21:
fixed, hopefully - I should refrain from committing stuff before properly testing it, my apologies!

closed 2016-11-21 by JakobEngel