JakobEngel/dso_ros #8: Issue compiling dso_ros

sevbest, 2016-12-06:
My dso lib is compiled with opencv as per the make command:
sev@sev-VirtualBox:~/code/dso/build$ cmake ..
-- Boost version: 1.58.0
-- Found the following Boost libraries:
--   system
--   thread
--   chrono
--   date_time
--   atomic
--- found PANGOLIN, compiling dso_pangolin library.
--- found OpenCV, compiling dso_opencv library.
--- found ziplib (1.1.1), compiling with zip capability.
--- compiling dso_dataset.
-- Configuring done
-- Generating done
-- Build files have been written to: /home/sev/code/dso/build

however rosmake is complaining about missing entrypoints in the dso lib:

[ 50%] Linking CXX executable bin/dso_live
  /home/sev/code/dso/build/lib/libdso.a(ImageDisplay_OpenCV.cpp.o): In function `dso::IOWrap::displayImage(char const*, cv::Mat const&, bool)':
  /home/sev/code/dso/src/IOWrapper/OpenCV/ImageDisplay_OpenCV.cpp:58: undefined reference to `cv::namedWindow(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, int)'
  /home/sev/code/dso/src/IOWrapper/OpenCV/ImageDisplay_OpenCV.cpp:59: undefined reference to `cv::resizeWindow(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, int, int)'
...

any idea what is going wrong here?

JakobEngel, 2016-12-06:
I believe that means the OpenCV libs are not linked correctly by rosmake (specifically the one that contains cv::namedWindow). 
I am however not sure how to tell ROS to properly link the openCV libs - the issue may also come from different OpenCV versions being linked / included by dso's cmake and rosmake - or different compiler versions (gcc 4.X vs 5.X) being used to build them.

archenroot, 2017-06-12:
Ok, this could be to me because OS OpenCV is different than one coming with ROS.

One question is, what version of OpenCV is DSO compatible with?

archenroot, 2017-06-14:
@sevbest - have you fixed this?

NikolausDemmel, 2017-11-29:
As Jakob wrote, mismatching compiler and/or opencv versions is the likely cause.

Can you provide some basic info: Which code version (both dso and dso_ros), what OS, which ROS version, which complier, and what are the outputs of cmake for dso and robuild for dso_ros.

Have you tried with catkin instead of rosbuild? There is a catkin branch in dso_ros.

burknator, 2017-11-30:
Oversaw that he wrote about different OpenCV versions. In fact, ROS seems to install it's own OpenCV 3.2, but libopencv-dev is version 2.4, which I used to compile dso with.

I think this is it, I will try a few things and get back to you.

burknator, 2017-11-30:
Yep, that's it. `sudo apt purge libopencv-dev` and then compiled dso again worked. It recognized the OpenCV package from ROS itself (`ros-kinetic-opencv`).
Also used the catkin branch and `catkin_make`.

NikolausDemmel, 2017-11-30:
Nice. Thanks for reporting back. I'll close this issue then.

closed 2017-11-30 by NikolausDemmel