JakobEngel/dso_ros #11: Issue with catkin_make

sunghoon031, 2016-12-22:
`catkin_make` gives the following result: 
````
[100%] Building CXX object dso_ros/CMakeFiles/dso_live.dir/src/main.o
Linking CXX executable /home/seonghunlee/ros/dso_ros/catkin_ws/devel/lib/dso_ros/dso_live
CMakeFiles/dso_live.dir/src/main.o: In function `main':
/home/seonghunlee/ros/dso_ros/catkin_ws/src/dso_ros/src/main.cpp:213: undefined reference to `dso::IOWrap::PangolinDSOViewer::PangolinDSOViewer(int, int, bool)'
collect2: error: ld returned 1 exit status
make[2]: *** [/home/seonghunlee/ros/dso_ros/catkin_ws/devel/lib/dso_ros/dso_live] Error 1
make[1]: *** [dso_ros/CMakeFiles/dso_live.dir/all] Error 2
make: *** [all] Error 2
Invoking "make -j4 -l4" failed
````

Can you maybe help me understand why I am having this error?

JakobEngel, 2016-12-25:
did you compile DSO with Pangolin?

sunghoon031, 2016-12-27:
@JakobEngel Yes. I compiled DSO with Pangolin and was able to run it successfully without any issue.

JakobEngel, 2017-01-02:
humm, yes, in that case Pangolin does not get linked correctly. Since I am unfamiliar with the catkin build system, unfortunately I cannot help you with that...

sunghoon031, 2017-02-28:
I found a dirty quick fix. I don't fully understand the implication of this, but it seems like it works.
I modified CMakeLists.txt as follows:
````
cmake_minimum_required(VERSION 2.4.6)
project(dso_ros)

find_package(catkin REQUIRED COMPONENTS
  geometry_msgs
  roscpp
  sensor_msgs
  cv_bridge
)

set(DSO_PATH $ENV{DSO_PATH})
message("\n\n---- LOOKING FOR DSO at\n\"${DSO_PATH}\"")

find_package(Pangolin 0.2 REQUIRED)
message("\n\n---- FOUND Pangolin Headers at\n\"${Pangolin_INCLUDE_DIRS}\"")
message("---- FOUND Pangolin Libs at\n\"${Pangolin_LIBRARIES}\"")

	message("--- found PANGOLIN, compiling dso_pangolin library.")
	set(dso_pangolin_SOURCE_FILES 
	  ${DSO_PATH}/src/IOWrapper/Pangolin/KeyFrameDisplay.cpp
	  ${DSO_PATH}/src/IOWrapper/Pangolin/PangolinDSOViewer.cpp)
	set(HAS_PANGOLIN 1)

find_package(OpenCV REQUIRED)
message("\n\n---- FOUND OpenCV Headers at\n\"${OpenCV_INCLUDE_DIRS}\"")
message("---- FOUND OpenCV Libs at\n\"${OpenCV_LIBS}\"")

    message("--- found OpenCV, compiling dso_opencv library.")
    include_directories( ${OpenCV_INCLUDE_DIRS} )
    set(dso_opencv_SOURCE_FILES 
      ${DSO_PATH}/src/IOWrapper/OpenCV/ImageDisplay_OpenCV.cpp
      ${DSO_PATH}/src/IOWrapper/OpenCV/ImageRW_OpenCV.cpp)
    set(HAS_OPENCV 1)

# find_package(Eigen3 REQUIRED)  ## This line is replaced by the following four lines
find_package(PkgConfig)
pkg_search_module(Eigen3 REQUIRED eigen3)
set(EIGEN3_INCLUDE_DIR "$ENV{EIGEN3_INCLUDE_DIR}" )
message("\n\n---- FOUND Eigen3 Headers at\n\"${EIGEN3_INCLUDE_DIR}\"")

find_package(Boost COMPONENTS system thread)
find_library(DSO_LIBRARY dso ${DSO_PATH}/build/lib)

catkin_package(
  CATKIN_DEPENDS
    geometry_msgs
    roscpp
    sensor_msgs
    cv_bridge
)

###########
## Build ##
###########

set(CMAKE_CXX_FLAGS
   "${SSE_FLAGS} -O3 -g -std=c++0x -march=native"
#   "${SSE_FLAGS} -O2 -g -std=c++0x -fno-omit-frame-pointer"
)

# Sources files
set(dso_SOURCE_FILES
  ${DSO_PATH}/src/FullSystem/FullSystem.cpp
  ${DSO_PATH}/src/FullSystem/FullSystemOptimize.cpp
  ${DSO_PATH}/src/FullSystem/FullSystemOptPoint.cpp
  ${DSO_PATH}/src/FullSystem/FullSystemDebugStuff.cpp
  ${DSO_PATH}/src/FullSystem/FullSystemMarginalize.cpp
  ${DSO_PATH}/src/FullSystem/Residuals.cpp
  ${DSO_PATH}/src/FullSystem/CoarseTracker.cpp
  ${DSO_PATH}/src/FullSystem/CoarseInitializer.cpp
  ${DSO_PATH}/src/FullSystem/ImmaturePoint.cpp
  ${DSO_PATH}/src/FullSystem/HessianBlocks.cpp
  ${DSO_PATH}/src/FullSystem/PixelSelector2.cpp
  ${DSO_PATH}/src/OptimizationBackend/EnergyFunctional.cpp
  ${DSO_PATH}/src/OptimizationBackend/AccumulatedTopHessian.cpp
  ${DSO_PATH}/src/OptimizationBackend/AccumulatedSCHessian.cpp
  ${DSO_PATH}/src/OptimizationBackend/EnergyFunctionalStructs.cpp
  ${DSO_PATH}/src/util/settings.cpp
  ${DSO_PATH}/src/util/Undistort.cpp
  ${DSO_PATH}/src/util/globalCalib.cpp
)

include_directories(
  ${PROJECT_SOURCE_DIR}/src
  ${DSO_PATH}/src
  ${DSO_PATH}/thirdparty/Sophus
  ${OpenCV_INCLUDE_DIRS}
  ${Pangolin_INCLUDE_DIRS}
  ${EIGEN3_INCLUDE_DIR}
  ${catkin_INCLUDE_DIRS}
)

include_directories( ${CSPARSE_INCLUDE_DIR} ${CHOLMOD_INCLUDE_DIR}) 
add_library(dso ${dso_SOURCE_FILES} ${dso_opencv_SOURCE_FILES} ${dso_pangolin_SOURCE_FILES})

## Declare a C++ executable
add_executable(dso_live src/main.cpp)

target_link_libraries(dso_live dso boost_system boost_thread cxsparse
    ${DSO_LIBRARY}
    ${Pangolin_LIBRARIES}
    ${OpenCV_LIBS}
    ${catkin_LIBRARIES})
````

closed 2017-10-24 by sunghoon031