JakobEngel/dso_ros #12: dso_live is not executable

melchiorharcourt, 2017-01-04:
dso is installed and works, dso_ros is compiled but it gives an error on execution :

rosrun dso_ros dso_live image=:cam1/image_raw \
calib=Datasets/EuroC/V1_01/camera.txt \
gamma=Datasets/EuroC/V1_01/pcalib.txt \
mode=1

[rosrun] Couldn't find executable named dso_live below /home/melchior/Workspace/modal/dso_ros
[rosrun] Found the following, but they're either not files,
[rosrun] or not executable:
[rosrun]   /home/melchior/Workspace/modal/dso_ros/build/bin/dso_live

Any idea ? The compilation worked properly (after having messed with path variables) and the file dso_live is in dso_ros/bin and weight 116Mo (is it usual ?)

Thank you !

Changliu52, 2017-01-05:
I have the same problem. Seems there was no dso_live executable generated after rosmake

Changliu52, 2017-01-05:
changliu@ubuntu:~/rosbuild_ws$ rosmake dso_ros 
[ rosmake ] rosmake starting...                                                 
[ rosmake ] Packages requested are: ['dso_ros']                                 
[ rosmake ] Logging to directory /home/changliu/.ros/rosmake/rosmake_output-20170105-202526
[ rosmake ] Expanded args ['dso_ros'] to:
['dso_ros']                           
[rosmake-0] Starting >>> catkin [ make ]                                        
[rosmake-0] Finished <<< catkin ROS_NOBUILD in package catkin
 No Makefile in package catkin
[rosmake-0] Starting >>> genmsg [ make ]                                        
[rosmake-0] Finished <<< genmsg ROS_NOBUILD in package genmsg
 No Makefile in package genmsg
[rosmake-0] Starting >>> genlisp [ make ]                                       
[rosmake-0] Finished <<< genlisp ROS_NOBUILD in package genlisp
 No Makefile in package genlisp
[rosmake-0] Starting >>> genpy [ make ]                                         
[rosmake-0] Finished <<< genpy ROS_NOBUILD in package genpy
 No Makefile in package genpy
[rosmake-0] Starting >>> gencpp [ make ]                                        
[rosmake-0] Finished <<< gencpp ROS_NOBUILD in package gencpp
 No Makefile in package gencpp
[rosmake-0] Starting >>> message_generation [ make ]                            
[rosmake-0] Finished <<< message_generation ROS_NOBUILD in package message_generation
 No Makefile in package message_generation
[rosmake-0] Starting >>> cmake_modules [ make ]                                 
[rosmake-0] Finished <<< cmake_modules ROS_NOBUILD in package cmake_modules
 No Makefile in package cmake_modules
[rosmake-0] Starting >>> rospack [ make ]                                       
[rosmake-1] Starting >>> cpp_common [ make ]                                    
[rosmake-1] Finished <<< cpp_common ROS_NOBUILD in package cpp_common
 No Makefile in package cpp_common
[rosmake-0] Finished <<< rospack ROS_NOBUILD in package rospack
 No Makefile in package rospack
[rosmake-1] Starting >>> rostime [ make ]                                       
[rosmake-0] Starting >>> roslib [ make ]                                        
[rosmake-1] Finished <<< rostime ROS_NOBUILD in package rostime
 No Makefile in package rostime
[rosmake-1] Starting >>> roscpp_traits [ make ]                                 
[rosmake-0] Finished <<< roslib ROS_NOBUILD in package roslib
 No Makefile in package roslib
[rosmake-0] Starting >>> rosunit [ make ]                                       
[rosmake-1] Finished <<< roscpp_traits ROS_NOBUILD in package roscpp_traits
 No Makefile in package roscpp_traits
[rosmake-1] Starting >>> roscpp_serialization [ make ]                          
[rosmake-0] Finished <<< rosunit ROS_NOBUILD in package rosunit
 No Makefile in package rosunit
[rosmake-0] Starting >>> roslang [ make ]                                       
[rosmake-0] Finished <<< roslang ROS_NOBUILD in package roslang
 No Makefile in package roslang
[rosmake-0] Starting >>> xmlrpcpp [ make ]                                      
[rosmake-0] Finished <<< xmlrpcpp ROS_NOBUILD in package xmlrpcpp
 No Makefile in package xmlrpcpp
[rosmake-0] Starting >>> rosgraph [ make ]                                      
[rosmake-0] Finished <<< rosgraph ROS_NOBUILD in package rosgraph
 No Makefile in package rosgraph
[rosmake-0] Starting >>> rosparam [ make ]                                      
[rosmake-1] Finished <<< roscpp_serialization ROS_NOBUILD in package roscpp_serialization
 No Makefile in package roscpp_serialization
[rosmake-1] Starting >>> message_runtime [ make ]                               
[rosmake-0] Finished <<< rosparam ROS_NOBUILD in package rosparam
 No Makefile in package rosparam
[rosmake-0] Starting >>> rosmaster [ make ]                                     
[rosmake-0] Finished <<< rosmaster ROS_NOBUILD in package rosmaster
 No Makefile in package rosmaster
[rosmake-0] Starting >>> rosclean [ make ]                                      
[rosmake-1] Finished <<< message_runtime ROS_NOBUILD in package message_runtime
 No Makefile in package message_runtime
[rosmake-0] Finished <<< rosclean ROS_NOBUILD in package rosclean
 No Makefile in package rosclean
[rosmake-1] Starting >>> std_msgs [ make ]                                      
[... 18 lines cut ...]
[rosmake-0] Starting >>> rosout [ make ]                                        
[rosmake-1] Finished <<< sensor_msgs ROS_NOBUILD in package sensor_msgs
 No Makefile in package sensor_msgs
[rosmake-1] Starting >>> rospy [ make ]                                         
[rosmake-0] Finished <<< rosout ROS_NOBUILD in package rosout
 No Makefile in package rosout
[rosmake-0] Starting >>> roslaunch [ make ]                                     
[rosmake-1] Finished <<< rospy ROS_NOBUILD in package rospy
 No Makefile in package rospy
[rosmake-0] Finished <<< roslaunch ROS_NOBUILD in package roslaunch
 No Makefile in package roslaunch
[rosmake-0] Starting >>> rostest [ make ]                                       
[rosmake-0] Finished <<< rostest ROS_NOBUILD in package rostest
 No Makefile in package rostest
[rosmake-0] Starting >>> cv_bridge [ make ]                                     
[rosmake-0] Finished <<< cv_bridge ROS_NOBUILD in package cv_bridge
 No Makefile in package cv_bridge
[rosmake-0] Starting >>> dso_ros [ make ]                                       
[rosmake-0] Finished <<< dso_ros [SKIP] No rule to make target None             
[ rosmake ] Results:                                                            
[ rosmake ] Built 34 packages with 0 failures.                                  
[ rosmake ] Summary output to directory                                         
[ rosmake ] /home/changliu/.ros/rosmake/rosmake_output-20170105-202526          
changliu@ubuntu:~/rosbuild_ws$ rosrun dso_ros dso_live
[rosrun] Couldn't find executable named dso_live below /home/changliu/rosbuild_ws/sandbox/dso_ros

lbxxdi, 2017-01-31:
i also found this problem and already fixed

edit file `dso_ros/CMakeLists.txt` 

from 
`set(EXECUTABLE_OUTPUT_PATH bin)`
to
`set(EXECUTABLE_OUTPUT_PATH ＄{PROJECT_SOURCE_DIR}/bin) `

closed 2017-02-20 by JakobEngel

vannem95, 2017-04-07:
I changed that but still getting the same error. @lbxxdi, did you do anything else other than this after doing **rosmake**? also, is dso_ros recognized as a package when you do **rospack list**, because it didn't for me. Thanks in advance.

lvxingzhe1991, 2017-06-02:
I changed that but still getting the same error. @vannem95 ,did you solve it?

aissa360, 2018-05-01:
@vannem95  Please how did you fix your problem ? because I am having the sale issue !!

NikolausDemmel, 2018-05-01:
I suggest using catkin with the respective branches, see also https://github.com/JakobEngel/dso_ros/issues/32#issuecomment-378578154

aissa360, 2018-05-04:
@NikolausDemmel Please I did what you said I cloned both the repositories in the src/ directory of my catkin_ws, I installed the dso using the cmake/make but I can't build the dso_ros, I am having the same issue as **@lorhm**  so I run git status :
```
On branch master
Your branch is up-to-date with 'origin/master'.
nothing to commit, working directory clean
```
and also git remote -v and here's what it gave :
```
origin	https://github.com/NikolausDemmel/dso_ros.git (fetch)
origin	https://github.com/NikolausDemmel/dso_ros.git (push)
```
Are these the your branches ?
P.S : I am using ubuntu 14.04 with ROS Indigo

NikolausDemmel, 2018-05-04:
Nope, those are not the right branches, see https://github.com/JakobEngel/dso_ros/issues/32#issuecomment-383882424

I haven't tried on 14.04 myself, but it should work I guess. My branch includes a travis configuration that compiles on trusty (https://github.com/NikolausDemmel/dso_ros/blob/catkin/.travis.yml) at least after upgrading the compiler.

NikolausDemmel, 2018-05-04:
You have to either install newer gcc, e.g. gcc5 and set the CC and CXX environment variables accordingly to make sure cmake choses that compile over the system default (4.8 I guess), or you can try to modify the CMakeLists.txt to use c++0x or c++11 instead of c++14 for both dso and dso_ros:

https://github.com/NikolausDemmel/dso/blob/5c77bcca3ca6b81abd6737f967f28442ec735317/CMakeLists.txt#L46

https://github.com/NikolausDemmel/dso_ros/blob/a54466c31adeebcb9a3d9c0a38c407d2bb0252f6/CMakeLists.txt#L89

aissa360, 2018-05-04:
Finally it worked ... thank you very much Mister @NikolausDemmel I really appreciate your HELP. 
the first thing is  in order the clone your specific branches I should add the -b [branch name] : 
`git clone https://github.com/JakobEngel/dso_ros.git -b cmake/catkin` 
Secondly I changed the  "c++14" to "c++11" in the two CMakeLists.txt files. 
Thank you again @NikolausDemmel  :)