JakobEngel/dso_ros #13: The Eigen error

cinnamennen, 2017-02-16:
I am trying to integrate this project with the ardrone_autonomy project, so that the video feed is fed into DSO. However, when trying to do so, it creates a core dump as follows:

```
loading calibration from /home/caleb/camera.txt!
NO GUI!
MODE WITH CALIBRATION, but without exposure times!
Reading Calibration from file /home/caleb/camera.txt ... found!
Input resolution: 640 360
In: 569.883158 568.007065 331.403348 135.879365 0.000000
Out: No Rectification
Output resolution: 640 360

Rectified Kamera Matrix:
569.883       0 331.403
      0 568.007 135.879
      0       0       1

NO PHOTOMETRIC Calibration!
Reading Photometric Calibration from file
PhotometricUndistorter: Could not open file!
using pyramid levels 0 to 3. coarsest resolution: 80 x 45!
dso_live: /usr/include/eigen3/Eigen/src/Core/DenseStorage.h:128: Eigen::internal::plain_array<T, Size, MatrixOrArrayOptions, 32>::plain_array() [with T = double; int Size = 4; int MatrixOrArrayOptions = 0]: Assertion `(reinterpret_cast<size_t>(eigen_unaligned_array_assert_workaround_gcc47(array)) & (31)) == 0 && "this assertion is explained here: " "http://eigen.tuxfamily.org/dox-devel/group__TopicUnalignedArrayAssert.html" " **** READ THIS WEB PAGE !!! ****"' failed.
[1]    53104 abort (core dumped)  rosrun dso_ros dso_live image:=/ardrone/image_raw calib=/home/caleb/camera.tx
```

I am currently using the devel branch of `dso` and the kinetic branch of `dso_ros`.

I have tracked this down to the `CalibHessian()` call in `HessianBlocks.h`, however I've not had much luck in continuing onwards. While I thought it could be an issue in `dso`, and not `dso_ros`, I am skeptical of this, as running it on sequence_01 appears to have correct functionality.

JirkaHarasim, 2017-02-18:
I have the identical problem. Dso sequences run, but dso_ros failes with the same dump.

How did you manage to run debugger with it? When I tried gdb, it didn't hop in and just ended.
rosrun --prefix 'gdb -ex run --args' dso_ros dso_live image:=\bebop\image_raw calib=/home/runtime/calibration_recordings/calibration/camera.txt

YuehChuan, 2017-02-20:
@cinnamennen  you need to compile dso (without ros version first), thanks to kgreenek
https://github.com/JakobEngel/dso/pull/37
then put camera.txt,  pcalib.txt vignette.png, at the right place

the git clone catkin branch version of dso_ros under your catkin_ws/src/

for running with webcam
for example 
`rosrun dso_ros dso_live image:=/usb_cam/image_raw calib=/home/rapyuta/dso_ws/src/dso_ros/camera.txt gamma=/home/rapyuta/dso_ws/src/dso_ros/pcalib.txt vignette=/home/rapyuta/dso_ws/src/dso_ros/vignette.png mode=1`

I not yet dive deeper to the code yet so not knowing the meaning of  pcalib.txt vignette.png
for camera.txt is about intrinsic matrix we get from opencv camera matrix
 fx fy cx cy d0 d1 d2 d3

It works for me!! 
 with ubuntu16.04 ROS Kinetic  eigen3  Good Luck!

JirkaHarasim, 2017-02-25:
Well you basically summed up the setup steps and said it works for you. Good for you :) 
A colleague of mine has the exactly same problem too. Do you have any idea what might be causing this?
To recap, pure dso works for me on example sequences, I compile dso_ros but running dso_live ends up with famous eigen assertion array problem. 
I didn't report it as an separate issue, because I believe it is the exact same problem as cinnamennen has.

cinnamennen, 2017-03-05:
@YuehChuan did you get the code to work with a live feed though? I've compiled with the fixed version of DSO, and I can run DSO_ROS on one of the TUM datasets, and that works. The problem arises as soon as I try to connect it to the image feed through ROS from ardrone_autonomy.

cinnamennen, 2017-03-06:
more specifically, the call out to the DSO library of ```fullSystem = new FullSystem();``` fails

YuehChuan, 2017-03-08:
@cinnamennen 
environment   intel i7 ROS kinetic, ubuntu16.04 opencv3.1
get it works with usb_cam feed 
dso
https://drive.google.com/file/d/0BwY-lpO6tzxHUzkzVnFTZDMzcU0/view?usp=sharing
dso with ROS
https://drive.google.com/file/d/0BwY-lpO6tzxHQzhxLW4tTkhUUXc/view?usp=sharing
By the way u need to have pangolin under your /home  path

FlorianFranzen, 2017-03-08:
As the error clearly states, one of the Eigen structures used is not properly aligned. For more details [please check their website](http://eigen.tuxfamily.org/dox-devel/group__TopicUnalignedArrayAssert.html).

If using any of the other Jakob Engel/TUM algorithms is any indicator, the cause is most likely that you are using a video resolution of 640x320 and not 640x480, which is what these algorithms were normally tested and developed on so far.

Never the less, without a stack trace of the crash, it will be impossible to say anything further. So if you want any help on this, compile dso with debugging symbols, crash it while running in a debugger and post the resulting stack trace here.

JirkaHarasim, 2017-03-12:
@YuehChuan  
The pangolin under /home path worked for me. Thanks man!

YuehChuan, 2017-03-13:
@JirkaHarasim  great! I'm glad that help :smile:

ChenFengYe, 2017-03-13:
Hi!   I got the same problem as your guys. And I move the whole pangolin folder from desktop into the /home path. Actually, I try the /home path and /home/user path twice. However, this bug is still there.
I have tested where this occurs. It happens Around Line 204.
 fullSystem = new FullSystem();
Same as @cinnamennen says. Anybody can help me?

vannem95, 2017-04-06:
@YuehChuan I am using your drive files to build my packages. when i try to rosrun it says the executable dso_live cannot be found. Can you provide a list of steps/commands you performed to get dso_ros to work? Thanks a lot in advance.

FlorianFranzen, 2017-04-06:
@vannem95 I had a similar issue, seems to be an inconsistency between ros versions in where binaries are placed. Just symlink the ```dso_ros``` binary in the ```build``` into the ```bin``` folder.

vannem95, 2017-04-07:
@FlorianFranzen there is no "dso_ros" in the build folder of my workspace. what did i do wrong?

vannem95, 2017-05-11:
How often does dso-ros crash for you guys? My pcalib and vignette are not specific to my camera. But it starts up and works fine as long as i don't move it too fast and as long as it is able to pick up points from the image.

fairf4x, 2017-05-12:
I am trying to debug rewrite of dso_ros wrapper from @lukx19 which should be able to publish _pointCloud_ and _odometry_ topics and I am stuck with the Eigen issue again.

I have uploaded [GDB backtraces ](https://github.com/fairf4x/dso_ros/issues/2) in my own fork od dso_ros.

I was not able to track the crash to some specific line of code.
If I set breakpoints to FullSystem constructor the program usually pass through it few times before the crash occurs. This makes it really elusive for any detailed tracking.
Any hints appreciated.

archenroot, 2017-06-13:
So what is the root cause here? is it related to dso issue, or eigen or pangolin versions involved?

fairf4x, 2017-06-13:
For me the solution/workaround was to disable eigen alignment in the source of [DSO](https://github.com/JakobEngel/dso/):
```
diff --git a/src/util/NumType.h b/src/util/NumType.h
index a12d420..0bdf2de 100644
--- a/src/util/NumType.h
+++ b/src/util/NumType.h
@@ -80,8 +80,8 @@ typedef Eigen::Matrix<float,CPARS,8> MatC8f;
 typedef Eigen::Matrix<double,8,8> Mat88;
 typedef Eigen::Matrix<double,7,7> Mat77;
 
-typedef Eigen::Matrix<double,CPARS,1> VecC;
-typedef Eigen::Matrix<float,CPARS,1> VecCf;
+typedef Eigen::Matrix<double,CPARS,1,Eigen::DontAlign> VecC;
+typedef Eigen::Matrix<float,CPARS,1,Eigen::DontAlign> VecCf;
 typedef Eigen::Matrix<double,13,1> Vec13;
 typedef Eigen::Matrix<double,10,1> Vec10;
 typedef Eigen::Matrix<double,9,1> Vec9;
```

archenroot, 2017-06-14:
@fairf4x that's my boy, going to test :-D thx!

NikolausDemmel, 2017-07-19:
So I think I might have tracked down this issue.

At least for me this was related to the fact that [DSO now uses 32 byte alignment](https://github.com/JakobEngel/dso/pull/37#issuecomment-279913488) (on Ubuntu 16.04 with according processor at least), by using the `-march=native` compile flag. At the same time dso_ros doesn't have this flag and thus Eigen uses 16 byte alignment. However the linked in code assumes 32 byte alignment, hence the crashes.

To fix it change the dso_ros `CMakeLists.txt` to have the same compile flags as dso, e.g.:

```
set(CMAKE_CXX_FLAGS "-std=c++14 -march=native" )
```

NikolausDemmel, 2017-07-19:
Just to be clear, even if you have worked around the crash in some other way, you should probably still make sure that the compile flags for dso and dso_ros are compatible. Otherwise you risk having all kinds of strange and hard-to-debug issues (possible candidate: https://github.com/JakobEngel/dso_ros/issues/21).

bzdfzfer, 2017-10-19:
I met the same problem and solved this by removing "-march=native" in the row 33 of CMakeLists.txt in dso project.

NikolausDemmel, 2017-10-19:
> I met the same problem and solved this by removing "-march=native" in the row 33 of CMakeLists.txt in dso project.

That also works. As mentioned, you need to make sure that the compile flags are the same or at least compatible, in particular with respect to alignment.

In any case I just merged adding `-march=native` to dso_ros

closed 2017-10-19 by NikolausDemmel

nunuwin, 2018-01-10:
I run dso_ros by @YuehChuan's file but I see like that, How to do do find dso_live? 

~$ rosrun dso_ros dso_live image:=/usb_cam/image_raw calib=/home/nu/dso_ws/src/dso_ros/camera.txt gamma=/home/nu/dso_ws/src/dso_ros/pcalib.txt vignette=/home/nu/dso_ws/src/dso_ros/vignette.png
[rosrun] Couldn't find executable named dso_live below /home/nu/catkin_ws/src/dso_ws/src/dso_ros

commit 47f2fb71 (2017-10-19): Merge pull request #24 from NikolausDemmel/patch-2

Add -march=native. Fixes #13.
commit fe26804f (2017-10-19): Add -march=native. Fixes #13.
commit 7b23d3f1 (2017-10-19): Add -march=native. Fixes #13.
commit cfb8fbb1 (2017-10-19): Merge pull request #23 from NikolausDemmel/patch-1

Add -march=native. Fixes #13.