JakobEngel/dso_ros #19: SIGABRT on >>  rosrun dso_ros dso_live image:=image_raw calib=src/dso_ros/live_camera_data/camera.txt gamma=src/dso_ros/live_camera_data/pcalib.txt vignette=src/dso_ros/live_camera_data/vignette.png

archenroot, 2017-06-13:
I have successfully upgraded to ROS Lunar and finished build of this project, but on rosrun exectuion I get:
```
zangetsu@ares ~/catkin_ws $ rosrun dso_ros dso_live image:=image_raw calib=src/dso_ros/live_camera_data/camera.txt gamma=src/dso_ros/live_camera_data/pcalib.txt vignette=src/dso_ros/live_camera_data/vignette.png 
loading calibration from src/dso_ros/live_camera_data/camera.txt!
loading gammaCalib from src/dso_ros/live_camera_data/pcalib.txt!
loading vignette from src/dso_ros/live_camera_data/vignette.png!
MODE WITH CALIBRATION, but without exposure times!
Reading Calibration from file src/dso_ros/live_camera_data/camera.txt ... found!
found ATAN camera model, building rectifier.
Creating FOV undistorter
Input resolution: 1280 1024
In: 0.535719 0.669567 0.493249 0.500409 0.897966

Found fx=0.535719, fy=0.669567, cx=0.493249, cy=0.500409.
 I'm assuming this is the "relative" calibration file format,and will rescale this by image width / height to fx=685.720714, fy=685.636463, cx=630.858138, cy=511.918472.

Out: Rectify Crop
Output resolution: 640 480
finding CROP optimal new model!
initial range: x: -1.1372 - 1.1901; y: -0.8311 - 0.8292!
iteration 00001: range: x: -1.1315 - 1.1841; y: -0.8311 - 0.8292!
iteration 00002: range: x: -1.1258 - 1.1782; y: -0.8311 - 0.8292!
iteration 00003: range: x: -1.1258 - 1.1782; y: -0.8270 - 0.8251!
iteration 00004: range: x: -1.1258 - 1.1782; y: -0.8228 - 0.8209!
iteration 00005: range: x: -1.1258 - 1.1782; y: -0.8228 - 0.8209!

Rectified Kamera Matrix:
 277.34       0 312.234
      0 291.402 239.777
      0       0       1

Reading Photometric Calibration from file src/dso_ros/live_camera_data/pcalib.txt
Reading Vignette Image from src/dso_ros/live_camera_data/vignette.png
Successfully read photometric calibration!
using pyramid levels 0 to 3. coarsest resolution: 80 x 60!
dso_live: /usr/include/eigen3/Eigen/src/Core/DenseStorage.h:128: Eigen::internal::plain_array<T, Size, MatrixOrArrayOptions, 32>::plain_array() [with T = double; int Size = 4; int MatrixOrArrayOptions = 0]: Assertion `(internal::UIntPtr(eigen_unaligned_array_assert_workaround_gcc47(array)) & (31)) == 0 && "this assertion is explained here: " "http://eigen.tuxfamily.org/dox-devel/group__TopicUnalignedArrayAssert.html" " **** READ THIS WEB PAGE !!! ****"' failed.
Neúspěšně ukončen (SIGABRT)

```

Not sure what to do, I have this EIGEN:
`dev-cpp/eigen-3.3.3`

maybe this version is not compatible?

archenroot, 2017-06-13:
Ok, so I would like to confirm that DSO itself works, I am just not able to run via ROS Lunar. Possible reason could be my change to update CMakeLists.txt file for Catkin workspace compatibility by following this guide:
http://wiki.ros.org/catkin/migrating_from_rosbuild

archenroot, 2017-06-13:
Additionally I tried to trace execution as debugger is not applicable here and see following:
```
rt_sigaction(SIGPIPE, {sa_handler=SIG_IGN, sa_mask=[PIPE], sa_flags=SA_RESTORER|SA_RESTART, sa_restorer=0x7fc63c25f290}, {sa_handler=SIG_DFL, sa_mask=[], sa_flags=0}, 8) = 0
uname({sysname="Linux", nodename="ares", ...}) = 0
mkdir("/home/zangetsu/.ros/", 0777)     = -1 EEXIST (File exists)
stat("/home/zangetsu/.ros/", {st_mode=S_IFDIR|0755, st_size=4096, ...}) = 0
mkdir("/home/zangetsu/.ros/log/", 0777) = -1 EEXIST (File exists)
stat("/home/zangetsu/.ros/log/", {st_mode=S_IFDIR|0755, st_size=4096, ...}) = 0
fstat(1, {st_mode=S_IFCHR|0620, st_rdev=makedev(136, 7), ...}) = 0
write(1, "loading calibration from src/dso"..., 66loading calibration from src/dso_ros/live_camera_data/pcalib.txt!
) = 66
write(1, "loading gammaCalib from src/dso_"..., 65loading gammaCalib from src/dso_ros/live_camera_data/camera.txt!
) = 65
write(1, "loading vignette from src/dso_ro"..., 65loading vignette from src/dso_ros/live_camera_data/vignette.png!
) = 65
write(1, "MODE WITH CALIBRATION, but witho"..., 51MODE WITH CALIBRATION, but without exposure times!
) = 51
open("src/dso_ros/live_camera_data/pcalib.txt", O_RDONLY) = 3
write(1, "Reading Calibration from file sr"..., 81Reading Calibration from file src/dso_ros/live_camera_data/pcalib.txt ... found!
) = 81
read(3, "0\t0.361816590639235\t0.4229556152"..., 8191) = 4331
close(3)                                = 0
write(1, "found RadTan (OpenCV) camera mod"..., 56found RadTan (OpenCV) camera model, building rectifier.
) = 56
write(1, "Creating RadTan undistorter\n", 28Creating RadTan undistorter
) = 28
open("src/dso_ros/live_camera_data/pcalib.txt", O_RDONLY) = 3
read(3, "0\t0.361816590639235\t0.4229556152"..., 8191) = 4331
read(3, "", 8191)                       = 0
write(1, "Failed to read camera calibratio"..., 52Failed to read camera calibration (invalid format?)
) = 52
write(1, "Calibration file: src/dso_ros/li"..., 58Calibration file: src/dso_ros/live_camera_data/pcalib.txt
) = 58
close(3)                                = 0
--- SIGSEGV {si_signo=SIGSEGV, si_code=SI_KERNEL, si_addr=NULL} ---
+++ killed by SIGSEGV +++
Neoprávněný přístup do paměti (SIGSEGV)

```

**Failed to read camera calibratio"..., 52Failed to read camera calibration (invalid format?**

NikolausDemmel, 2017-06-13:
Have you tried the catkin branch of this repo?

archenroot, 2017-06-13:
@NikolausDemmel - ah, I wasn't aware, thx will take a look.

archenroot, 2017-06-14:
@NikolausDemmel I have resolved the compile time even on Master branch by applying following patch:
```
diff --git a/src/util/NumType.h b/src/util/NumType.h
index a12d420..0bdf2de 100644
--- a/src/util/NumType.h
+++ b/src/util/NumType.h
@@ -80,8 +80,8 @@ typedef Eigen::Matrix<float,CPARS,8> MatC8f;
 typedef Eigen::Matrix<double,8,8> Mat88;
 typedef Eigen::Matrix<double,7,7> Mat77;
 
-typedef Eigen::Matrix<double,CPARS,1> VecC;
-typedef Eigen::Matrix<float,CPARS,1> VecCf;
+typedef Eigen::Matrix<double,CPARS,1,Eigen::DontAlign> VecC;
+typedef Eigen::Matrix<float,CPARS,1,Eigen::DontAlign> VecCf;
 typedef Eigen::Matrix<double,13,1> Vec13;
 typedef Eigen::Matrix<double,10,1> Vec10;
 typedef Eigen::Matrix<double,9,1> Vec9;
```

closed 2017-06-14 by archenroot