JakobEngel/dso_ros #20: DSO_ROS with web cam - is supported?

archenroot, 2017-06-14:
Hi, 
first thanks for this project, it is amazing!

Maybe it is better question at ROS forum, but when I run:
```
rosrun \
dso_ros \
dso_live \
image:=/camera/image_raw \
video_device:=/dev/video0 \
calib=src/dso_ros/live_camera_data/camera.txt \
gamma=src/dso_ros/live_camera_data/pcalib.txt \
vignette=src/dso_ros/live_camera_data/vignette.png
```

The application starts with following output:
```
loading calibration from src/dso_ros/live_camera_data/camera.txt!
loading gammaCalib from src/dso_ros/live_camera_data/pcalib.txt!
loading vignette from src/dso_ros/live_camera_data/vignette.png!
MODE WITH CALIBRATION, but without exposure times!
Reading Calibration from file src/dso_ros/live_camera_data/camera.txt ... found!
found ATAN camera model, building rectifier.
Creating FOV undistorter
Input resolution: 1280 1024
In: 0.535719 0.669567 0.493249 0.500409 0.897966

Found fx=0.535719, fy=0.669567, cx=0.493249, cy=0.500409.
 I'm assuming this is the "relative" calibration file format,and will rescale this by image width / height to fx=685.720714, fy=685.636463, cx=630.858138, cy=511.918472.

Out: Rectify Crop
Output resolution: 640 480
finding CROP optimal new model!
initial range: x: -1.1372 - 1.1901; y: -0.8311 - 0.8292!
iteration 00001: range: x: -1.1315 - 1.1841; y: -0.8311 - 0.8292!
iteration 00002: range: x: -1.1258 - 1.1782; y: -0.8311 - 0.8292!
iteration 00003: range: x: -1.1258 - 1.1782; y: -0.8270 - 0.8251!
iteration 00004: range: x: -1.1258 - 1.1782; y: -0.8228 - 0.8209!
iteration 00005: range: x: -1.1258 - 1.1782; y: -0.8228 - 0.8209!

Rectified Kamera Matrix:
 277.34       0 312.234
      0 291.402 239.777
      0       0       1

Reading Photometric Calibration from file src/dso_ros/live_camera_data/pcalib.txt
Reading Vignette Image from src/dso_ros/live_camera_data/vignette.png
Successfully read photometric calibration!
using pyramid levels 0 to 3. coarsest resolution: 80 x 60!
START PANGOLIN!
```
But the screen remain completely BLACK, any idea? I don't have in the moment other than standard laptop integrated webcam...

Thanks.

archenroot, 2017-06-14:
I am using Mono set..

Mosquito19, 2017-07-05:
I have the same problem ,when I run :dso_ros dso_live image:=/usb/cam/image_raw calib=/home/zhao/camera.txt mode=1;
 it out put:
loading calibration from /home/zhao/camera.txt!
could not parse argument "mode=1"!!
MODE WITH CALIBRATION, but without exposure times!
Reading Calibration from file /home/zhao/camera.txt ... found!
found RadTan (OpenCV) camera model, building rectifier.
Creating RadTan undistorter
Input resolution: 640 480
In: 446.867338 446.958766 298.082779 234.334299 -0.324849 0.120516 -0.000186 -0.000821
Out: Rectify Crop
Output resolution: 640 480
finding CROP optimal new model!
initial range: x: -0.7956 - 0.9601; y: -0.5852 - 0.6177!
iteration 00001: range: x: -0.7916 - 0.9553; y: -0.5852 - 0.6177!
iteration 00002: range: x: -0.7876 - 0.9505; y: -0.5852 - 0.6177!
iteration 00003: range: x: -0.7876 - 0.9505; y: -0.5823 - 0.6146!
iteration 00004: range: x: -0.7876 - 0.9505; y: -0.5794 - 0.6116!
iteration 00005: range: x: -0.7876 - 0.9505; y: -0.5794 - 0.6116!

Rectified Kamera Matrix:
367.628       0 289.559
      0 402.213 233.025
      0       0       1

NO PHOTOMETRIC Calibration!
Reading Photometric Calibration from file 
PhotometricUndistorter: Could not open file!
using pyramid levels 0 to 3. coarsest resolution: 80 x 60!
START PANGOLIN!

do you solve your problem? could you help me ?

bobaner, 2018-01-08:
I got the almost identical problem, could anyone help plz

rosrun dso_ros dso_live image:=/camera_node/image_raw calib=~/catkin_ws/src/dso_ros/camera.txt  mode=1
loading calibration from /home/bohui/catkin_ws/src/dso_ros/camera.txt!
PHOTOMETRIC MODE WITHOUT CALIBRATION!
Reading Calibration from file /home/bohui/catkin_ws/src/dso_ros/camera.txt ... found!
found RadTan (OpenCV) camera model, building rectifier.
Creating RadTan undistorter
Input resolution: 640 480
In: 446.867338 446.958766 298.082779 234.334299 -0.324849 0.120516 -0.000186 -0.000821
Out: Rectify Crop
Output resolution: 640 480
finding CROP optimal new model!
initial range: x: -0.7956 - 0.9601; y: -0.5852 - 0.6177!
iteration 00001: range: x: -0.7916 - 0.9553; y: -0.5852 - 0.6177!
iteration 00002: range: x: -0.7876 - 0.9505; y: -0.5852 - 0.6177!
iteration 00003: range: x: -0.7876 - 0.9505; y: -0.5823 - 0.6146!
iteration 00004: range: x: -0.7876 - 0.9505; y: -0.5794 - 0.6116!
iteration 00005: range: x: -0.7876 - 0.9505; y: -0.5794 - 0.6116!

Rectified Kamera Matrix:
367.628       0 289.559
      0 402.213 233.025
      0       0       1

NO PHOTOMETRIC Calibration!
Reading Photometric Calibration from file 
PhotometricUndistorter: Could not open file!
using pyramid levels 0 to 3. coarsest resolution: 80 x 60!
dso_live: /usr/local/include/eigen3/Eigen/src/Core/DenseStorage.h:128: Eigen::internal::plain_array<T, Size, MatrixOrArrayOptions, 32>::plain_array() [with T = double; int Size = 4; int MatrixOrArrayOptions = 0]: Assertion `(internal::UIntPtr(eigen_unaligned_array_assert_workaround_gcc47(array)) & (31)) == 0 && "this assertion is explained here: " "http://eigen.tuxfamily.org/dox-devel/group__TopicUnalignedArrayAssert.html" " **** READ THIS WEB PAGE !!! ****"' failed.
Aborted (core dumped)

NikolausDemmel, 2018-01-09:
Can someone provide a small bag and calibration file with which to reproduce this?

farhanmubasher, 2018-05-09:
hi @archenroot ,
I think your DSO is not getting any kind of pictures so that it is showing black screen, You should check whether your camera is active or not. You will have to publish your real time camera coordinates using CV Bridge on a topic "raw_image" so that DSO can have that coordinates and can do further processing.

Pedrous, 2018-06-19:
@archenroot ,
The same thing happened to me and after searching for a long time I noticed that I was trying to interface a topic that did not exist. So I would advice you to check that you are trying to subscribe to the correct topic and also that this topic is publishing the images as you expect.

YuCodes, 2018-10-10:
i got the same error when trying to run it. do i have to publish the camera topic separately