JakobEngel/dso_ros #22: When run dso_live , the  screen has no picture,no vedio,just black ?

Mosquito19, 2017-07-05:
I use a usb camera; 
when I run :dso_ros dso_live image:=/usb/cam/image_raw calib=/home/zhao/camera.txt mode=1;
it out put:
loading calibration from /home/zhao/camera.txt!
could not parse argument "mode=1"!!
MODE WITH CALIBRATION, but without exposure times!
Reading Calibration from file /home/zhao/camera.txt ... found!
found RadTan (OpenCV) camera model, building rectifier.
Creating RadTan undistorter
Input resolution: 640 480
In: 446.867338 446.958766 298.082779 234.334299 -0.324849 0.120516 -0.000186 -0.000821
Out: Rectify Crop
Output resolution: 640 480
finding CROP optimal new model!
initial range: x: -0.7956 - 0.9601; y: -0.5852 - 0.6177!
iteration 00001: range: x: -0.7916 - 0.9553; y: -0.5852 - 0.6177!
iteration 00002: range: x: -0.7876 - 0.9505; y: -0.5852 - 0.6177!
iteration 00003: range: x: -0.7876 - 0.9505; y: -0.5823 - 0.6146!
iteration 00004: range: x: -0.7876 - 0.9505; y: -0.5794 - 0.6116!
iteration 00005: range: x: -0.7876 - 0.9505; y: -0.5794 - 0.6116!

Rectified Kamera Matrix:
367.628 0 289.559
0 402.213 233.025
0 0 1

NO PHOTOMETRIC Calibration!
Reading Photometric Calibration from file
PhotometricUndistorter: Could not open file!
using pyramid levels 0 to 3. coarsest resolution: 80 x 60!
START PANGOLIN!
```

 The  interface  has no picture,no vedio,just black.
I don't know why...help ~~thanks !!!

Mosquito19, 2017-07-05:
I solved !! I write a wrong sentence :run :dso_ros dso_live image:=/usb/cam/image_raw calib=/home/zhao/camera.txt mode=1;------------change to :dso_ros dso_live image:=/usb_cam/image_raw calib=/home/zhao/camera.txt mode=1

NikolausDemmel, 2017-07-24:
@Mosquito19, maybe you can close the issue?

closed 2017-10-09 by Mosquito19