JakobEngel/dso_ros #25: Error: double free or corruption

giorgiovaccarino, 2017-10-18:
I try to use dso_ros with rosbag EuRoC dataset MH_01.
Ubuntu 16.04.3 64 bit.
ROS Kinetic

I received the error below.
Dso itself is working.
My camera.txt is:
458.654	457.296	367.215 248.375	0.0
752 480
none
752 480

I omitted gamma and vignette.
May the error be related to wrong camera parameter or bad expressed (as its work on ORB_SLAM2 ROS node)? Or is another issue?
Does anyone experimented this?
Thank you.

-----------------------------------------------------------------
gvaccarino@ubuntu:~/dso/build$ rosrun dso_ros dso_live image:=image_raw calib=/home/gvaccarino/datasets/MH_01/dsoconfig/camera.txt
loading calibration from /home/gvaccarino/datasets/MH_01/dsoconfig/camera.txt!
MODE WITH CALIBRATION, but without exposure times!
Reading Calibration from file /home/gvaccarino/datasets/MH_01/dsoconfig/camera.txt ... found!
found PINHOLE camera model, building rectifier.
Input resolution: 752 480
In: 458.654000 457.296000 367.215000 248.375000 0.000000
Out: No Rectification
Output resolution: 752 480

Rectified Kamera Matrix:
458.654       0 367.215
      0 457.296 248.375
      0       0       1

NO PHOTOMETRIC Calibration!
Reading Photometric Calibration from file 
PhotometricUndistorter: Could not open file!
using pyramid levels 0 to 4. coarsest resolution: 47 x 30!
START PANGOLIN!
Framebuffer with requested attributes not available. Using available framebuffer. You may see visual artifacts.*** Error in `/home/gvaccarino/dso_ros/bin/dso_live': double free or corruption (out): 0x0000000001168240 ***
======= Backtrace: =========
/lib/x86_64-linux-gnu/libc.so.6(+0x777e5)[0x7f8a503637e5]
/lib/x86_64-linux-gnu/libc.so.6(+0x8037a)[0x7f8a5036c37a]
/lib/x86_64-linux-gnu/libc.so.6(cfree+0x4c)[0x7f8a5037053c]
/home/gvaccarino/dso_ros/bin/dso_live(_Z5vidCbN5boost10shared_ptrIKN11sensor_msgs6Image_ISaIvEEEEE+0x265)[0x438f85]
/home/gvaccarino/dso_ros/bin/dso_live(_ZN5boost6detail8function22void_function_invoker1IPFvNS_10shared_ptrIKN11sensor_msgs6Image_ISaIvEEEEEEvS9_E6invokeERNS1_15function_bufferES9_+0x3d)[0x439f9d]
/home/gvaccarino/dso_ros/bin/dso_live(_ZN3ros27SubscriptionCallbackHelperTIN5boost10shared_ptrIKN11sensor_msgs6Image_ISaIvEEEEEvE4callERNS_36SubscriptionCallbackHelperCallParamsE+0x2e1)[0x443041]
/opt/ros/kinetic/lib/libroscpp.so(_ZN3ros17SubscriptionQueue4callEv+0xadd)[0x7f8a5391e5dd]
/opt/ros/kinetic/lib/libroscpp.so(_ZN3ros13CallbackQueue9callOneCBEPNS0_3TLSE+0x4a0)[0x7f8a538c8cf0]
/opt/ros/kinetic/lib/libroscpp.so(_ZN3ros13CallbackQueue13callAvailableENS_12WallDurationE+0x5b3)[0x7f8a538ca0f3]
/opt/ros/kinetic/lib/libroscpp.so(_ZN3ros21SingleThreadedSpinner4spinEPNS_13CallbackQueueE+0x2a1)[0x7f8a539226a1]
/opt/ros/kinetic/lib/libroscpp.so(_ZN3ros4spinEv+0x2b)[0x7f8a5390773b]
/home/gvaccarino/dso_ros/bin/dso_live(main+0x429)[0x436c89]
/lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf0)[0x7f8a5030c830]
/home/gvaccarino/dso_ros/bin/dso_live(_start+0x29)[0x438739]

NikolausDemmel, 2017-10-23:
You could check if you are affected by an issue caused by Eigen with different compile options (see https://github.com/JakobEngel/dso_ros/issues/13).

Best pull the latest version of DSO and dso_ros and recompile everything from scratch with the updated compile flags.

giorgiovaccarino, 2017-10-24:
What I did:
I delete dso and dso_ros from disk.
I remove Eigen source code I had compiled in the past.
I pull the latest version of dso and dso_ros.
Install Eigen with sudo apt-get install libsuitesparse-dev libeigen3-dev libboost-all-dev.
Compiled dso and dso_ros.
In dso_ros remains open the question about path of bin ([rosrun] Couldn't find executable named dso_live below /home/gvaccarino/dso_ros). I moved manually dso_live from: dso_ros/build/bin to: dso_ros/bin.
Then I ran:
t1: roscore
t2: rosbag play --pause ~/datasets/MH_01/MH_01_easy.bag /cam0/image_raw:=/image_raw
t3: rosrun dso_ros dso_live image:=image_raw calib=/home/gvaccarino/datasets/MH_01/dsoconfig/camera.txt

In brief, when I start the rosbag, the error is the same. 
Does this error depend on Ubuntu installation in VMWare?
Can there be any problems with the settings in camera.txt?

Here the result:
loading calibration from /home/gvaccarino/datasets/MH_01/dsoconfig/camera.txt!
MODE WITH CALIBRATION, but without exposure times!
Reading Calibration from file /home/gvaccarino/datasets/MH_01/dsoconfig/camera.txt ... found!
found PINHOLE camera model, building rectifier.
Input resolution: 752 480
In: 458.654000 457.296000 367.215000 248.375000 0.000000
Out: No Rectification
Output resolution: 752 480

Rectified Kamera Matrix:
458.654       0 367.215
      0 457.296 248.375
      0       0       1

NO PHOTOMETRIC Calibration!
Reading Photometric Calibration from file 
PhotometricUndistorter: Could not open file!
using pyramid levels 0 to 4. coarsest resolution: 47 x 30!
START PANGOLIN!
Framebuffer with requested attributes not available. Using available framebuffer. You may see visual artifacts.*** Error in `/home/gvaccarino/dso_ros/bin/dso_live': double free or corruption (out): 0x00000000014e5fa0 ***
======= Backtrace: =========
/lib/x86_64-linux-gnu/libc.so.6(+0x777e5)[0x7f273e8147e5]
/lib/x86_64-linux-gnu/libc.so.6(+0x8037a)[0x7f273e81d37a]
/lib/x86_64-linux-gnu/libc.so.6(cfree+0x4c)[0x7f273e82153c]
/home/gvaccarino/dso_ros/bin/dso_live(_Z5vidCbN5boost10shared_ptrIKN11sensor_msgs6Image_ISaIvEEEEE+0x24d)[0x438fdd]
/home/gvaccarino/dso_ros/bin/dso_live(_ZN5boost6detail8function22void_function_invoker1IPFvNS_10shared_ptrIKN11sensor_msgs6Image_ISaIvEEEEEEvS9_E6invokeERNS1_15function_bufferES9_+0x41)[0x43a151]
/home/gvaccarino/dso_ros/bin/dso_live(_ZN3ros27SubscriptionCallbackHelperTIN5boost10shared_ptrIKN11sensor_msgs6Image_ISaIvEEEEEvE4callERNS_36SubscriptionCallbackHelperCallParamsE+0x2f6)[0x4435f6]
/opt/ros/kinetic/lib/libroscpp.so(_ZN3ros17SubscriptionQueue4callEv+0xadd)[0x7f2741dcf5dd]
/opt/ros/kinetic/lib/libroscpp.so(_ZN3ros13CallbackQueue9callOneCBEPNS0_3TLSE+0x4a0)[0x7f2741d79cf0]
/opt/ros/kinetic/lib/libroscpp.so(_ZN3ros13CallbackQueue13callAvailableENS_12WallDurationE+0x5b3)[0x7f2741d7b0f3]
/opt/ros/kinetic/lib/libroscpp.so(_ZN3ros21SingleThreadedSpinner4spinEPNS_13CallbackQueueE+0x2a1)[0x7f2741dd36a1]
/opt/ros/kinetic/lib/libroscpp.so(_ZN3ros4spinEv+0x2b)[0x7f2741db873b]
/home/gvaccarino/dso_ros/bin/dso_live(main+0x43c)[0x436cac]
/lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf0)[0x7f273e7bd830]
/home/gvaccarino/dso_ros/bin/dso_live(_start+0x29)[0x438769]

NikolausDemmel, 2017-10-24:
Which branches of dso and dso ros do you use? How do you compile dso_ros?

I can try to reproduce with the same dataset.

giorgiovaccarino, 2017-10-24:
I pull the master branch with:
git clone git@github.com:JakobEngel/dso.git
git clone git@github.com:JakobEngel/dso_ros.git

I compiled dso with:
cd dso
mkdir build
cd build
cmake ..
make -j

I compiled dso_ros with:
cd dso_ros
export DSO_PATH=~/dso
rosmake

I have just finished a test to compile the projects removing any switches (but without a real knowledge about it).
For dso I  remove -march=native and for dso_ros -march=native and -fno-omit-frame-pointer.
The UI (Pangolin) now starts but in initial stage this error occours: 
gvaccarino@ubuntu:~/dso_ros$ rosrun dso_ros dso_live image:=image_raw calib=/home/gvaccarino/datasets/MH_01/dsoconfig/camera.txt
loading calibration from /home/gvaccarino/datasets/MH_01/dsoconfig/camera.txt!
MODE WITH CALIBRATION, but without exposure times!
Reading Calibration from file /home/gvaccarino/datasets/MH_01/dsoconfig/camera.txt ... found!
found PINHOLE camera model, building rectifier.
Input resolution: 752 480
In: 458.654000 457.296000 367.215000 248.375000 0.000000
Out: No Rectification
Output resolution: 752 480

Rectified Kamera Matrix:
458.654       0 367.215
      0 457.296 248.375
      0       0       1

NO PHOTOMETRIC Calibration!
Reading Photometric Calibration from file 
PhotometricUndistorter: Could not open file!
using pyramid levels 0 to 4. coarsest resolution: 47 x 30!
START PANGOLIN!
Framebuffer with requested attributes not available. Using available framebuffer. You may see visual artifacts.Initialization: keep 11.1% (need 1200, have 10785)!
INITIALIZE FROM INITIALIZER (1231 pts)!
SPARSITY:  MinActDist 2.100000 (need 1200 points, have 1231 points)!
OPTIMIZE 1231 pts, 1231 active res, 0 lin res!
Initial Error       	A(593933.713867)=(AV inf). Num: A(0) + M(0); ab 0.000000 0.000000!
STEPS: A 14.7; B 57.9; R 2729.1; T 44109.5. 	REJECT 0 (L -1.00, dir nan, ss 1.0): 	A(4929670.708160)=(AV 453.214). Num: A(3) + M(0); ab -0.124555 4.916804!
STEPS: A 45.0; B 379.8; R 48.8; T 8441.2. 	ACCEPT 1 (L -1.60, dir -0.59, ss 1.0): 	A(4816824.393547)=(AV 31.891). Num: A(592) + M(0); ab -0.506350 37.144749!
STEPS: A 49.3; B 381.3; R 58.5; T 0.4. 	ACCEPT 2 (L -2.20, dir 0.07, ss 1.0): 	A(4577883.751007)=(AV 28.251). Num: A(717) + M(0); ab -0.924354 69.495628!
STEPS: A 36.2; B 148.3; R 58.0; T 1.1. 	REJECT 3 (L -2.81, dir 0.96, ss 1.0): 	A(4580168.015907)=(AV 28.743). Num: A(693) + M(0); ab -1.231346 82.081535!
STEPS: A 27.9; B 45.4; R 87.0; T 2.2. 	ACCEPT 4 (L -3.41, dir 0.95, ss 1.0): 	A(4272851.523760)=(AV 27.389). Num: A(712) + M(0); ab -1.468141 85.931168!
STEPS: A 21.3; B 1.0; R 22.8; T 0.8. 	ACCEPT 5 (L -4.01, dir 0.91, ss 1.0): 	A(3855215.316185)=(AV 25.782). Num: A(725) + M(0); ab -1.649291 85.843307!
STEPS: A 18.2; B 8.0; R 9.6; T 0.4. 	ACCEPT 6 (L -4.61, dir 0.99, ss 1.0): 	A(3482227.719835)=(AV 24.335). Num: A(735) + M(0); ab -1.803516 85.161591!
STEPS: A 15.3; B 8.7; R 17.6; T 0.3. 	ACCEPT 7 (L -5.21, dir 0.99, ss 1.0): 	A(3206216.925194)=(AV 23.241). Num: A(742) + M(0); ab -1.933350 84.420914!
STEPS: A 13.3; B 6.0; R 14.9; T 0.2. 	ACCEPT 8 (L -5.82, dir 1.00, ss 1.0): 	A(2949685.161999)=(AV 22.202). Num: A(748) + M(0); ab -2.045979 83.908768!
STEPS: A 11.5; B 4.6; R 15.9; T 0.3. 	ACCEPT 9 (L -6.42, dir 1.00, ss 1.0): 	A(2752080.234953)=(AV 21.374). Num: A(753) + M(0); ab -2.143786 83.514343!
STEPS: A 10.7; B 1.2; R 8.1; T 0.1. 	ACCEPT 10 (L -7.02, dir 1.00, ss 1.0): A(2643941.935215)=(AV 20.908). Num: A(756) + M(0); ab -2.234362 83.409073!
STEPS: A 10.3; B 1.1; R 12.4; T 0.3. 	ACCEPT 11 (L -7.62, dir 1.00, ss 1.0): A(2556376.487243)=(AV 20.559). Num: A(756) + M(0); ab -2.321532 83.314117!
STEPS: A 9.5; B 2.3; R 5.2; T 0.1. 	ACCEPT 12 (L -8.22, dir 0.99, ss 1.0): A(2485709.263422)=(AV 20.300). Num: A(754) + M(0); ab -2.402453 83.115089!
STEPS: A 9.4; B 0.7; R 10.1; T 0.3. 	ACCEPT 13 (L -8.83, dir 1.00, ss 1.0): A(2433440.101053)=(AV 20.112). Num: A(752) + M(0); ab -2.482378 83.059074!
STEPS: A 8.8; B 2.9; R 6.8; T 0.1. 	ACCEPT 14 (L -9.43, dir 1.00, ss 1.0): A(2341826.845357)=(AV 19.691). Num: A(755) + M(0); ab -2.557351 82.816124!
LOG 7: 19.450 fine. Res: 755 A, 0 L, 0 M; (0 / 0) forceDrop. a=-2.557351, b=82.816126. Window 7 (2)
INCREASING cutoff to 40.000000 (ratio is 0.448598)!
REPEAT LEVEL!
BIG ERROR! tracking failed entirely. Take predictred pose and hope we may somehow recover.
Coarse Tracker tracked ab = -2.557351 82.816126 (exp 1.000000). Res nan!
Initial Tracking failed: LOST!
^CQUIT Pangolin thread!
I'll just kill the whole process.
So Long, and Thanks for All the Fish!

NikolausDemmel, 2017-10-24:
Hm... maybe the error with the compile flags is somehow related to running in a virtual machine. I have never tried that, so no idea...

The output you are seeing now means that tracking fails. IIRC the paper even mentions that DSO has trouble on those sequences, in particular the sequence start. I think you can try starting the sequence not at the beginning, but somewhere in between.

giorgiovaccarino, 2017-10-24:
I will try soon playing the sequence with differents start points.
I will try also, in a few days, to test the dos_ros node on a native Ubuntu system (and finally in Gazebo environment).
I will report the results.

giorgiovaccarino, 2017-10-26:
I tried with native Ubuntu pc and it works.
May be the virtualized environment interferes with compiler optimization.
The slam fails in initialization stage of EuroC sequence but can be played by skipping this part.

NikolausDemmel, 2017-10-26:
Hm... strange indeed.

giorgiovaccarino, 2017-11-06:
I correct myself. I forgot to correct the cmakelists.txt file again after download a fresh source copy.
The previous was modified from me:
#set(EXECUTABLE_OUTPUT_PATH bin)
set(EXECUTABLE_OUTPUT_PATH ${PROJECT_SOURCE_DIR}/bin)
So I always test the older bin with the older compiler switches.
Now, after restored these lines, the dso ros node works (also in Gazebo).

closed 2017-11-06 by NikolausDemmel