JakobEngel/dso_ros #28: Producing Webcam calibration files

yikuangteh, 2018-01-04:
Hi there

I'm relatively new to this, so I'm hoping if there is anyone out there who could help me out.
I understand that I am supposed to calibrate my camera using whatever algorithms out there to do so (Eg. OpenCV to calibrate the camera). But I'm completely lost as to how the values for the Vignette and Gamma are created and calibrated for the camera. Is there a simple way to do so?

Thank you!

NikolausDemmel, 2018-01-04:
First of all, you can use DSO without calibration. Check out `mode=1` in the [DSO readme](https://github.com/JakobEngel/dso). For the `dso_ros` example, mode 0 is hardcoded, but you change that to your needs. The results should be worse than with proper calibration, but it should work.

Also, check out "Photometric Calibration" in the same Readme to find out about how you can do that.

Lastly, your webcam is likely rolling shutter. Keep in mind that DSO is designed for global shutter cameras. Depending on how fast the line readout times are, this will effect the performance negatively as you move the camera faster. For slow movements it should still work fine though.

yikuangteh, 2018-01-05:
@NikolausDemmel Okay, thank you so much!

closed 2018-01-05 by yikuangteh

az-ja, 2018-01-12:
@NikolausDemmel Can you please explain a bit, how to change the code of `dso_ros` to run it for mode=1? (No photometric calibration)
Is that a small changing of some variables or I should copy some code from `dso` and paste there? Or is it more complicated than that? :( 
I changed line 192 in the `main` of `dso_ros` to `setting_photometricCalibration = 0;` as in `dso` that was set when the mode was set to 1.
when I run `./dso_live image:=/cv_camera/image_raw calib=/home/calib.txt` I get this error:

loading calibration from /home/calib.txt!
MODE WITH CALIBRATION, but without exposure times!
Reading Calibration from file /home/calib.txt ... found!
Input resolution: 640 480
In: 654.750000 653.680000 307.610000 223.260000 0.000000
Out: Rectify Full
Output resolution: 640 480
dso_live: /home/azin/dso/dso/src/util/Undistort.cpp:713: void dso::Undistort::makeOptimalK_full(): Assertion `false' failed.
Aborted (core dumped)
 :(
Thanks for any help.

Turing47, 2023-01-29:
> @NikolausDemmel Can you please explain a bit, how to change the code of `dso_ros` to run it for mode=1? (No photometric calibration) Is that a small changing of some variables or I should copy some [...]

I checked the code. If you choose "full" mode, the algorithm stops working since the corresponding function is not completed.