JakobEngel/dso_ros #29: Retrieving/Streaming Output Data

yikuangteh, 2018-01-11:
Hi

I would like to stream out the data that is being printed from the function given in the sampleOutputWrapper (pushDepthImageFloat) into, let's say, a Python script.
Is there a way to do so? I tried to find the function in the Main and System codes but to no avail.

NikolausDemmel, 2018-01-11:
You can add whatever output code you want in the output wrapper callback, so you can e.g. open a tcp connection or a Unix pipe and serialize the data, and then connect and deserialize in your python script.

Maybe one or the simple options is using ROS. Usw dso_ros and add an output wrapper where you publish the data you want in the respective callbacks. On the python side you have subscribers listening to those topics.

yikuangteh, 2018-01-12:
Okay, thank you!
Is it possible to get the data (Eg. Depth and coordinates of points) from the plotted graph in Pangolin and send the data as well? Do I do that within the OutputWrapper or within the programming file for Pangolin?

yikuangteh, 2018-01-12:
I happened to see this post in the main DSO Github: https://github.com/JakobEngel/dso/issues/55

I couldn't make out how the values are related to what the camera is seeing other than the depth value on the last column. But I guess it's somewhat the thing that I needed to produce an output outside of Pangolin. Is it possible to output them as a data stream into a Python file? @ginko3

@NikolausDemmel I also understand that you've advised not to do any sort of processing in the OutputWrapper file, but is it still possible to do so? (Eg. Determining a depth value and passing commands to the ROS based how the value it gives)

NikolausDemmel, 2018-08-15:
> I also understand that you've advised not to do any sort of processing in the OutputWrapper file, but is it still possible to do so? (Eg. Determining a depth value and passing commands to the ROS based on the value it gives)

Yes, just that if you want real-time operation, you shouldn't do any long processing. If you don't care about realtime and run in "linearized" execution, it doesn't matter. Otherwise, what you can / should do is putting whatever data you need in some queue, and then have another thread processing the data from the queue and doing whatever you want.