JakobEngel/dso_ros #32: Unable to Build dso_ros

lorhm, 2018-04-03:
[CMakeOutput.log](https://github.com/JakobEngel/dso_ros/files/1871584/CMakeOutput.log)

I followed the instructions in the README.md.

I ran:
` export DSO_PATH=/home/rahulr/Workspace/dso`
` rosmake`
But i Got:

`[ rosmake ] rosmake starting...                                                  `
`[ rosmake ] No package or stack specified.  And current directory 'build' is not a package name or stack name.`
`[ rosmake ] Packages requested are: []                                             `
`[ rosmake ] Logging to directory /home/rahulr/.ros/rosmake/rosmake_output-20180403-184032`
`[ rosmake ] Expanded args [] to:`
`[]                                                `
`[ rosmake ] ERROR: No arguments could be parsed into valid package or stack names. `

--------------------------------------------------------------------
I when I tried to create the "build" folder and run 
` cmake .. `

I got: 

`[rosbuild] Building package dso_ros`
`[rosbuild] Error from directory check: /opt/ros/kinetic/share/ros/core/rosbuild/bin/check_same_directories.py  /home/rahulr/Workspace/dso_ros
1`
`Traceback (most recent call last):`
`  File "/opt/ros/kinetic/share/ros/core/rosbuild/bin/check_same_directories.py", line 46, in <module>
    raise Exception`
`Exception`
`CMake Error at /opt/ros/kinetic/share/ros/core/rosbuild/private.cmake:102 (message):`
`  [rosbuild] rospack found package "dso_ros" at "", but the current directory
  is "/home/rahulr/Workspace/dso_ros".  You should double-check your
  ROS_PACKAGE_PATH to ensure that packages are found in the correct
  precedence order.`
`Call Stack (most recent call first):
  /opt/ros/kinetic/share/ros/core/rosbuild/public.cmake:177 (_rosbuild_check_package_location)
  CMakeLists.txt:13 (rosbuild_init)`

`-- Configuring incomplete, errors occurred!`
`See also "/home/rahulr/Workspace/dso_ros/build/CMakeFiles/CMakeOutput.log".`

I am trying to get the dso to run on a live camera feed - either from the laptop camera or an externally attached usb cam. At this point I'm unsure of how to proceed.
Any help would be appreciated, Thanks!

NikolausDemmel, 2018-04-04:
I suggest to use `catkin` instead of `rosbuild`, with the `catkin` branch of this repository.

Even better, I suggest to use my branches from the PR's https://github.com/JakobEngel/dso/pull/109 and https://github.com/JakobEngel/dso_ros/pull/27, which hopefully fix some build system issues:
 * https://github.com/NikolausDemmel/dso/tree/cmake
 * https://github.com/NikolausDemmel/dso_ros/tree/catkin

SteveJos, 2018-04-10:
Hey @NikolausDemmel ...is there any aditional changes needed in order to compile the dso wrapper? I am cloning your repo to the `src` folder of my ROS workspace and it even isn recognized by `catkin_make`?
Could you please provide further steps? 
Thank you in advance!

NikolausDemmel, 2018-04-10:
Have you checked out the cmake branch for dso and catkin branch for dso_ros?

NikolausDemmel, 2018-04-10:
BTW I'm using catkin tools, with that, you can also put dso in your catkin workspace and dso_ros should find it. Otherwise, you might need to specify the location of dso in the environment variable for dso_ros to find it.

lorhm, 2018-04-13:
@SteveJos and @NikolausDemmel ,
Once I clone the catkin version of DSO_ROS into the src of catkin_ws, How do I build the files. I am very new to ROS and have little to no clue how to proceed.

lorhm, 2018-04-23:
@NikolausDemmel - I cloned the two repositories you had linked to the catkin_ws/src.
In catkin_ws/src/dso I ran - 
 mkdir build
cd build
cmake ..
make -j

This built DSO in the catkin_ws. I cloned the catkin version of the DSO_ROS into  catkin_ws/src/

I am facing two doubts. How do I build the dso_ros package using catkintools/catkin_make? 
or 
Should i go about the same procedure of creaking build folder, running cmake and make?

NikolausDemmel, 2018-04-23:
With ROS packages, it is probably best to not manually call cmake and make. I suggest using `catkin_tools`. Assuming you have all dependencies installed, and both repositories at the specified branches checked out into `catkin_ws/src/`, you can build with:

```
# make sure you have the setup.bash for the base ROS install sourced (assuming Kinetic here)
source /opt/ros/kinetic/setup.bash

cd catkin_ws
catkin init
catkin config -DCMAKE_BUILD_TYPE=Release
catkin build
```

lorhm, 2018-04-24:
Hi Nikolaus, Thank you for your reply.

I followed what you had suggested, but the dso_ros did not get built.
First I cleaned the catkin workspace **catkin_ws** using the command:
`catkin clean --deinit
`
I cloned the both the repositories into the **catkin_ws/src** 
I compiled dso using the standard cmake technique as suggested in the README.md with opencv.
Then I proceeded to follow the steps you had mentioned in your answer to build **dso_ros** 

But I still find that dso_ros was not built.
I ran **catkin list**
This did not show dso_ros as one of the packages in the catkin_ws.

Any suggestions would be great. Thank you for replying once again!

NikolausDemmel, 2018-04-24:
Please run both `git status` and `git remote -v` in the two folders `dso` and `dso_ros` inside your workspace and post the output here.

NikolausDemmel, 2018-04-24:
You need to use the branches mentioned above, not master. Go checkout `cmake` for dso and `catkin` for dso_ros.

lorhm, 2018-04-24:
Nikolaus, you legend it compiled! Can't believe That was the issue.. Its compiling and working.

One last query with two sub-queries ( :D ) -
1) Does dso_ros run directly from the laptop camera? 
2) what should I do to configure it to work from an external camera source, if I want to run this on a micro controller with a peripheral camera?

lorhm, 2018-04-25:
Im able to compile it now Nikolaus. Thank you. But I am getting 
`[rospack] Error: package 'dso_ros' not found`

dso_ros was compiled and it did 

```
$ catkin list
- dso_ros
- rqt_svo
- svo_analysis
- svo_msgs
- vikit_common
- vikit_py
- vikit_ros
- svo
- svo_ros

```

NikolausDemmel, 2018-05-01:
> [ERROR] [1524752132.229356642]: [registerPublisher] Failed to contact master at [localhost:11311]. Retrying...

You will have to start `roscore` in a different terminal before running `dso_live`.

NikolausDemmel, 2018-05-01:
Please close the issue if it is resolved. Thanks!

lorhm, 2018-05-03:
Thanks Nikolaus!

closed 2018-05-03 by lorhm

NikolausDemmel, 2018-05-06:
> 1. Does dso_ros run directly from the laptop camera?

> 2. what should I do to configure it to work from an external camera source, if I want to run this on a micro controller with a peripheral camera?

To work with live cameras, you will have to read up on how that works with ROS. ROS has a bunch of camera drivers for various devices: http://wiki.ros.org/Sensors/Cameras

Also, keep in mind that DSO will probably not work very well on a webcam, at least not for faster motion, since it assumes global shutter and most webcams are rolling shutter.

MIL35, 2019-06-10:
@NikolausDemmel, @lorhm,  
I've followed all your instructions, as I was facing the same issue. When I `catkin build` in my `catkin_ws/src` now, my `dso` package succesfully builds, but my `dso_ros` package fails with the error 
`CMakeFiles/dso_live.dir/src/main.cpp.o: In function `main':
main.cpp:(.text.startup+0x5b9): undefined reference to `dso::IOWrap::PangolinDSOViewer::PangolinDSOViewer(int, int, bool)'
`.
I've tried the modified `CMakeLists.txt` in the issue [https://github.com/JakobEngel/dso_ros/issues/11](url), but it hasn't solved the problem.

Further, my VS Code editor flags the following lines in the `main.cpp` of `dso_ros` as `cannot open source file`:

'#include "util/settings.h"
#include "FullSystem/FullSystem.h"
#include "util/Undistort.h"
#include "IOWrapper/Pangolin/PangolinDSOViewer.h"
#include "IOWrapper/OutputWrapper/SampleOutputWrapper.h"
'
I am not able to figure out what to change or modify such that the `main.cpp` can see these files and use the functions in them.
Any guidance would be appreciated, Thanks!

bigbellmercy, 2019-09-25:
> With ROS packages, it is probably best to not manually call cmake and make. I suggest using `catkin_tools`. Assuming you have all dependencies installed, and both repositories at the specified [...]

How to use 'catkin_make' to build 'dso_live':

As an another method to build, I succeeded in build by 'catkin_make' in kinetic ROS, instead of catkin_tools above.
> export DSO_PATH="here write the path of 'dso' folder"
> catkin_make

After that, since the dso_live package is not recognized, the following command should be done once as mentioned above post.
> source XXX/catkin_ws/devel/setup.zsh