JakobEngel/dso_ros #33: How to implement real-time DSO-SLAM on Windows

lcqwe, 2018-04-29:
Anyone have ever managed to add some real-time-streaming function (just like from a mono camera) on Winows using VS2015 or other compiler until now? I recently tried to implement this, but I don't know how to adapt this DSO_ROS to Windows(as we all know there is not ROS env here) . Where I need to add code into, sampleoutput or some other place? Hope can give me some prompt if you know about this.
Thanks in advance.

NikolausDemmel, 2018-08-15:
The minimal thing you need to do is feeding images into DSO, like https://github.com/JakobEngel/dso_ros/blob/cfb8fbb167fb0a0ae45ff46bf7b9b64258ac32bc/src/main.cpp#L160-L162

lcqwe, 2018-08-18:
already solved. Thank you

closed 2018-08-18 by lcqwe