JakobEngel/dso_ros #34: Build map using dso_ros and reuse it for autonomous purpose

farhanmubasher, 2018-05-09:
Hi all..
I'm using dso_ros for real time tracking and i'm getting poses from camToWorld.data() ( e.g x,y,z, qx, qy, qz, qw). But how can i build map using these poses and reuse that map.
For exmaple, I want to move a robot in a square. So firstly i will move it manually in a square so that it can build a map and after that it should move in that map autonomously .

is there any other way to do it ? (Build map from unknown environment using dso and then reuse it so robot can move autonomously in that involvement).
Thanks in advance !!

NikolausDemmel, 2018-08-15:
DSO is "only" a keyframe odometry. There is no persistent map so far.

NikolausDemmel, 2018-10-04:
We have released LDSO, which integrates loop closure detection and pose graph optimization. This is still not quite what you need for map re-use. The missing parts are storing a map and releasing it and registering a newly initialized odometry against the old map (could be done with the same loop closure detection). Eventually one also would need some map management / keyframe pruning for long term operation, but all the building blocks should be there.

Here is the link: https://github.com/tum-vision/LDSO

I'll close this issue for now.

closed 2018-10-04 by NikolausDemmel