JakobEngel/dso_ros #36: how to run dso_ros with multiple cameras

farhanmubasher, 2018-06-22:
Hi all,

Currently i'm using single camera and running DSO using dso_ros for real time, but what if i have multiple cameras, how can i run dso_ros with multiple cameras

Thanks

NikolausDemmel, 2018-06-22:
dso & dso_ros only supports a monocular setup.

farhanmubasher, 2018-06-22:
Thanks for your reply. I got it
What if i want to use multiple cameras with DSO to run it for better accuracy ?
And how stereo cameras works dso if it supports only monocular setup ?

NikolausDemmel, 2018-06-22:
The code in this repo and https://github.com/JakobEngel/dso is only for a single camera. If you have a stereo setup, you can only use either camera.

I'm aware of two different (independent) stereo extensions for DSO. One is https://github.com/JiatianWu/stereo-dso and the other one is https://vision.in.tum.de/research/vslam/stereo-dso. For the latter no code is publicly available unfortunately.

farhanmubasher, 2018-06-22:
Ok it means right now we can't run dso with multiple or stereo cameras in real time .
can you please tell me why ros is necessary to run dso ? can we skip it and use any other kernel or any other thing to run dso for real time ?

closed 2018-06-22 by farhanmubasher

NikolausDemmel, 2018-07-05:
> can you please tell me why ros is necessary to run dso ? can we skip it and use any other kernel or any other thing to run dso for real time ?

Yes, if you implement a corresponding OutputWrapper yourself (take the ROS version as an example).