JakobEngel/dso_ros #38: variable DSO_LIBRARY not found

squashking, 2018-08-15:
Hi guys, I downloaded the catkin branch. I am trying to compile it using catkin_make (ROS Lunar), and I got the following error:

CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
DSO_LIBRARY
    linked by target "dso_live" in directory /******/catkin_ws/src/dso_ros-catkin

Could you give me any suggestions? Thank you very much!

NikolausDemmel, 2018-08-15:
It means the dso library wasn't found in `$DSO_PATH/build/lib`. https://github.com/JakobEngel/dso_ros/blob/47f2fb718e95125c7afd20989dd57d35c3234d5f/CMakeLists.txt#L25

Make sure you set `DSO_PATH` correct and that the dso build can be found in the according subdirectory.

Otherwise, I generally recommend to use the following branches from my fork:

- branch `catkin` from https://github.com/NikolausDemmel/dso_ros
- branch `cmake` from https://github.com/NikolausDemmel/dso

If you drop both in a catkin workspace, you should be able to build them with `catkin_tools` (i.e. `catkin build` command) without specifying any paths manually (make sure to unset `DSO_PATH` if you have already set it permanently).

squashking, 2018-08-15:
@NikolausDemmel  Thank you very much Nikolaus. Sorry to ask a newbie question, but how to set DSO_PATH ?

NikolausDemmel, 2018-08-15:
Before calling `cmake`, in the terminal type something like

```
export DSO_PATH=/absolute/path/to/your/dso/directory
```

Remember it expects `build/lib` subdirectory inside `DSO_PATH` and also `src` and `thirdparty/Sophus`.

squashking, 2018-08-15:
@NikolausDemmel Thanks for the reply. I am using catkin_make not cmake, and I see there is only src subdirectory. Shall I add the above subdirectories as you mentioned?

NikolausDemmel, 2018-08-15:
`DSO_PATH` is supposed to be the directory of `dso` and the `build` directory inside needs to be the directory where u ran cmake and make to compile `dso`.

squashking, 2018-08-15:
@NikolausDemmel Sorry I'm not using the cmake branch, I'm using the catkin branch: https://github.com/NikolausDemmel/dso_ros/tree/catkin 
As I'm a newbie, I only know how to compile it using catkin_make. After downloading the code, the is no build/lib subdirectories, so I created them. And then export DSO_LIBRARY=/absolute/path/to/the/lib/folder. And then I just used the catkin_make command. Did I do something wrong? Thank you!

NikolausDemmel, 2018-08-16:
`DSO_PATH` (not `DSO_LIBRARY`) needs to point to the `dso` (not `dso_ros`) directory which you got from https://github.com/JakobEngel/dso. This you need to build with cmake inside a subfolder called `build` as per it's readme. You cannot use catkin_make to build `dso`. Then you can build `dso_ros` with catkin_make.

I'm sorry, but I seriously suggest you first do some basic cmake and catkin_make tutorials to learn the basic workflow of these tools. This will make things much easier.

squashking, 2018-08-16:
@NikolausDemmel Thanks, I was confused at the beginning. Now I understand the two branches are correlated. I have installed DSO using cmake. But when trying the dataset (sequence02) from your website I got the error message and no visualization dialogue displaying:

eading Photometric Calibration from file ../../sequence_02/pcalib.txt
Reading Vignette Image from ../../vignette.png
cv::imread could not read image ../../vignette.png! this may segfault. 
cv::imread could not read image ../../vignette.png! this may segfault. 
PhotometricUndistorter: Invalid vignette image
got 3500 images and 3500 timestamps and 3500 exposures.!
ImageFolderReader: got 3500 files in ../../sequence_02/images.zip!
using pyramid levels 0 to 3. coarsest resolution: 80 x 60!
 
Any suggestions? Thank you!

NikolausDemmel, 2018-08-16:
You are running DSO now without dos_ros? For one thing, looks like the path to the vignette is wrong.

squashking, 2018-08-16:
@NikolausDemmel Yes, I'm running DSO alone. The path does not seem to be wrong and I can see the vignette image file. Also, if you look at the first part of the log, the images and calibration files are found:

loading data from ../../sequence_02/images.zip!
loading calibration from ../../sequence_02/camera.txt!
loading gammaCalib from ../../sequence_02/pcalib.txt!
loading vignette from ../../vignette.png!

=============== PRESET Settings: ===============
DEFAULT settings:
- no  real-time enforcing
- 2000 active points
- 5-7 active frames
- 1-6 LM iteration each KF
- original image resolution
==============================================
PHOTOMETRIC MODE WITH CALIBRATION!
got 3500 entries and 3500 files!
Reading Calibration from file ../../sequence_02/camera.txt ... found!
found ATAN camera model, building rectifier.
Creating FOV undistorter
Input resolution: 1280 1024
In: 0.349153 0.436593 0.493140 0.499021 0.933271

So I don't know what went wrong here.

NikolausDemmel, 2018-08-17:
I don't know. You'll have to debug into the code and see which path it is taking...

squashking, 2018-08-21:
Thank you Nikolaus, It turns out I forgot to add /sequence_02 to the vignette path...sorry about the mistake.

closed 2018-08-21 by squashking