JakobEngel/dso_ros #39: Catkin branch gives the errors about "librealsense.so"

zhouyan09, 2018-09-04:
Environment: Ubuntu14.04, ros: indigo version
Issue: When I git download the catkin branch, using command "catkin_make", it gives the errors below:
"/opt/ros/indigo/lib/librealsense.so: undefined reference to `libusb_strerror'
/opt/ros/indigo/lib/librealsense.so: undefined reference to `libusb_get_port_numbers'
/opt/ros/indigo/lib/librealsense.so: undefined reference to `libusb_get_parent'
/opt/ros/indigo/lib/librealsense.so: undefined reference to `libusb_error_name'
collect2: error: ld returned 1 exit status
"

///////////////////////////////////////////////////////////////////////////////////////////
In the directory /opt/ros/indigo/lib/, input the command "ls -l | grep librealsense", the results are :
"
lrwxrwxrwx 1 root root      17  March 30  2017 librealsense.so -> librealsense.so.1
lrwxrwxrwx 1 root root      22  March 30  2017 librealsense.so.1 -> librealsense.so.1.12.1
-rw-r--r-- 1 root root  959216  March 30  2017 librealsense.so.1.12.1
"

///////////////////////////////////////////////////////////////////////////////////////////

I try to reinstall the library using command "sudo apt-get --reinstall install 'ros-indigo-librealsense'", but is does not work, can anyone help me solve this problem? Thanks a lot.

NikolausDemmel, 2018-09-04:
Sound like an error nothing to with dso_ros. Looks like the apt package for librealsense is outdated. Have you updated all apt packages? If yes, seems to be an upstream issue.

Neither DSO nor DSO_ROS use librealsense directly AFAIK. It might come in via pangolin though. You might want to try to compile librealsense yourself (e.g. http://wiki.ros.org/librealsense?distro=indigo in your catkin workspace) and then recompiling pangolin making sure it uses that.

Or maybe better, disable librealsense when compiling pangolin.

NikolausDemmel, 2018-09-05:
If both DSO and DSO_ROS link the same librealsense.so, and one works but the other doesn't, that is strange. Maybe you need to manually link some libusb for DSO_ROS, which seems to happen for DSO.

1. Use ldtree to check the linked libraries for both DSO and DSO_ROS. See if they link the same pangolin lib and the same librealsense.
2. If yes, check if DSO maybe has some additional library like libusb or something that is missing for DSO_ROS.

Otherwise, I don't know...