JakobEngel/dso_ros #43: could not parse argument "mode=1"!!

robotlee1997, 2019-01-18:
I try to run dso_ros, but I can't run and see "could not parse argument "mode=2"!!".
$ rosrun dso_ros dso_live image:=/usb_cam/image_raw calib=～/cam.txt mode=1
loading calibration from ～/cam.txt!
could not parse argument "mode=1"!!
MODE WITH CALIBRATION, but without exposure times!
Reading Calibration from file ～/cam.txt ... not found. Cannot operate without calibration, shutting down.
Segmentation fault (core dumped)

NikolausDemmel, 2019-01-18:
I don't think parsing `mode` is implemented for `dso_ros` (search https://github.com/JakobEngel/dso_ros/blob/master/src/main.cpp), since you never get exposure times with the ROS topic. You have to change it yourself in the code if you need it.

robotlee1997, 2019-01-22:
> I don't think parsing `mode` is implemented for `dso_ros` (search https://github.com/JakobEngel/dso_ros/blob/master/src/main.cpp), since you never get exposure times with the ROS topic. You have to change it yourself in the code if you need it.

Maybe my computer has something wrong. I reinstall dso_ros, then this error is gone.

robotlee1997, 2019-01-22:
> I don't think parsing `mode` is implemented for `dso_ros` (search https://github.com/JakobEngel/dso_ros/blob/master/src/main.cpp), since you never get exposure times with the ROS topic. You have to change it yourself in the code if you need it.

thank you for your replay.

closed 2019-01-22 by robotlee1997