JakobEngel/dso_ros #48: Installation and documentation about dso_ros?

ibrahimLearning, 2020-06-10:
Thanks for your nice repo. I successfully installed dso from your repo: https://github.com/JakobEngel/dso

Now I would like to run the dso in the ros framework. But the documentation makes me some confusion. According to your documentation, after typing:
 export DSO_PATH=[PATH_TO_DSO]/dso
 rosmake

Will I build dso_ros again or not?

robert-meiner, 2020-06-14:
I asked this question also in #47.  No answers, so far. 

It would be good to know how this was set up and built by the @JakobEngel and some of the other people who are using it.

An alternative method of building with Catkintools is addressed by @NikolausDemmel in #32

jotix16, 2020-10-28:
@ibrahimLearning  @robert-meiner any solution?

IldarGreat, 2023-03-11:
Maybe it help https://github.com/IldarGreat/dso/tree/ros_catkin