JakobEngel/dso_ros #49: Installing on Raspberry Pi4b+

robert-meiner, 2020-06-16:
Dear Nikolaus,  in issue #32, you advise @rahul95ram:

"You need to use the branches mentioned above, not master. Go checkout `cmake` for dso and `catkin` for dso_ros."

_Originally posted by @NikolausDemmel in https://github.com/JakobEngel/dso_ros/issues/32#issuecomment-383882424_

I am not able to use the dso branch you mention, since I am building on raspberry.  I used code posted by @chabaudmorgan "Compilation for Raspberry Pi 3 #80" https://github.com/JakobEngel/dso/pull/80/files.  I changed the -mtune value on line 34 of that code to -mtune=cortex-a72, to reflect what is on the pi4b+

This code is built in my catkin_ws/src/dso folder.  It compiled and works fine with the TUM sample databases.

Then I cloned the catkin branch of dso_ros into catkin_ws/src.  When I attempt to follow the instructions that worked for @rahul95ram, no package is found or built.  Here is what I get:

pi@antonio:~ $ source /opt/ros/melodic/setup.bash
pi@antonio:~ $ cd catkin_ws
pi@antonio:~/catkin_ws $ catkin init
Catkin workspace `/home/pi/catkin_ws` is already initialized. No action taken.
-------------------------------------------------------
Profile:                     default
Extending:             [env] /opt/ros/melodic
Workspace:                   /home/pi/catkin_ws
-------------------------------------------------------
Build Space:       [missing] /home/pi/catkin_ws/build
Devel Space:       [missing] /home/pi/catkin_ws/devel
Install Space:      [unused] /home/pi/catkin_ws/install
Log Space:          [exists] /home/pi/catkin_ws/logs
Source Space:       [exists] /home/pi/catkin_ws/src
DESTDIR:            [unused] None
-------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
-------------------------------------------------------
Additional CMake Args:       -DCMAKE_BUILD_TYPE=Release
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
-------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
-------------------------------------------------------
Workspace configuration appears valid.
-------------------------------------------------------
pi@antonio:~/catkin_ws $ catkin config -DCMAKE_BUILD_TYPE=Release
-------------------------------------------------------
Profile:                     default
Extending:             [env] /opt/ros/melodic
Workspace:                   /home/pi/catkin_ws
-------------------------------------------------------
Build Space:       [missing] /home/pi/catkin_ws/build
Devel Space:       [missing] /home/pi/catkin_ws/devel
Install Space:      [unused] /home/pi/catkin_ws/install
Log Space:          [exists] /home/pi/catkin_ws/logs
Source Space:       [exists] /home/pi/catkin_ws/src
DESTDIR:            [unused] None
-------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
-------------------------------------------------------
Additional CMake Args:       -DCMAKE_BUILD_TYPE=Release
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
-------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
-------------------------------------------------------
Workspace configuration appears valid.
-------------------------------------------------------
pi@antonio:~/catkin_ws $ catkin build
-------------------------------------------------------
Profile:                     default
Extending:             [env] /opt/ros/melodic
Workspace:                   /home/pi/catkin_ws
-------------------------------------------------------
Build Space:        [exists] /home/pi/catkin_ws/build
Devel Space:        [exists] /home/pi/catkin_ws/devel
Install Space:      [unused] /home/pi/catkin_ws/install
Log Space:          [exists] /home/pi/catkin_ws/logs
Source Space:       [exists] /home/pi/catkin_ws/src
DESTDIR:            [unused] None
-------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
-------------------------------------------------------
Additional CMake Args:       -DCMAKE_BUILD_TYPE=Release
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
-------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
-------------------------------------------------------
Workspace configuration appears valid.
-------------------------------------------------------
[build] No packages were found in the source space '/home/pi/catkin_ws/src'
[build] No packages to be built.
[build] Package table is up to date.                                           
Warning: generated devel space setup files have been deleted.
Starting  >>> catkin_tools_prebuild                                            
Finished  <<< catkin_tools_prebuild                [ 9.8 seconds ]             
[build] Summary: All 1 packages succeeded!                                     
[build]   Ignored:   None.                                                     
[build]   Warnings:  None.                                                     
[build]   Abandoned: None.                                                     
[build]   Failed:    None.                                                     
[build] Runtime: 9.8 seconds total.                                            
pi@antonio:~/catkin_ws $ 

the git status of the catkin branch of dso_ros checks out.

pi@antonio:~/catkin_ws/src $ cd dso_ros
pi@antonio:~/catkin_ws/src/dso_ros $ git status
On branch master
Your branch is up to date with 'origin/master'.

nothing to commit, working tree clean
pi@antonio:~/catkin_ws/src/dso_ros $ git remote -v
origin	https://github.com/NikolausDemmel/dso_ros.git (fetch)
origin	https://github.com/NikolausDemmel/dso_ros.git (push)
pi@antonio:~/catkin_ws/src/dso_ros $ 

the git status of dso won't check out, since I built this from some locally modified code.  But dso seems to work, so is this important?

pi@antonio:~/catkin_ws/src/dso $ git status
On branch master
Your branch is up to date with 'origin/master'.

Changes not staged for commit:
  (use "git add <file>..." to update what will be committed)
  (use "git checkout -- <file>..." to discard changes in working directory)

	modified:   CMakeLists.txt

Untracked files:
  (use "git add <file>..." to include in what will be committed)

	CMakeLists_old.txt

no changes added to commit (use "git add" and/or "git commit -a")
pi@antonio:~/catkin_ws/src/dso $ git remote -v
origin	https://github.com/JakobEngel/dso.git (fetch)
origin	https://github.com/JakobEngel/dso.git (push)
pi@antonio:~/catkin_ws/src/dso $ 

Best Regards,
Mark

NikolausDemmel, 2020-06-17:
Sorry, I don't have the time to take a closer look. I haven't used this in a long time, so I don't know how out-of-date it is. You could try with the cmake branch of DSO and apply additional modifications you need for your platform manually from your other branch.

The main DSO branch should also work somehow. One difference I think is that there you need to set the DSO path variable for dso-ros to find it, IIRC.

robert-meiner, 2020-06-17:
Dear Nikolaus, 

Thank you for the quick reply and sorry to pull you back into the past.  Your first paragraph describes what I have done with dso.  My dso package works fine, based on successful running of the TUM sample datasets.

The problem is with the dso_ros wrapper.  Since I wrote this, I have installed another ros node package in the workspace using catkin tools, so catkin tools and ros seem to be functioning properly.

When I try to install the dso_ros wrapper, no package is built.

I have also tried this in the following way, with no success:

pi@antonio:~/catkin_ws $ source /opt/ros/melodic/setup.bash
pi@antonio:~/catkin_ws $ export DSO_PATH=~/catkin_ws/src/dso
pi@antonio:~/catkin_ws $ catkin init
Catkin workspace `/home/pi/catkin_ws` is already initialized. No action taken.
-------------------------------------------------------
Profile:                     default
Extending:          [cached] /opt/ros/melodic
Workspace:                   /home/pi/catkin_ws
-------------------------------------------------------
Build Space:        [exists] /home/pi/catkin_ws/build
Devel Space:        [exists] /home/pi/catkin_ws/devel
Install Space:      [unused] /home/pi/catkin_ws/install
Log Space:          [exists] /home/pi/catkin_ws/logs
Source Space:       [exists] /home/pi/catkin_ws/src
DESTDIR:            [unused] None
-------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
-------------------------------------------------------
Additional CMake Args:       -DCMAKE_BUILD_TYPE=Release
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
-------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
-------------------------------------------------------
Workspace configuration appears valid.
-------------------------------------------------------
pi@antonio:~/catkin_ws $ catkin config -DCMAKE_BUILD_TYPE=Release
-------------------------------------------------------
Profile:                     default
Extending:          [cached] /opt/ros/melodic
Workspace:                   /home/pi/catkin_ws
-------------------------------------------------------
Build Space:        [exists] /home/pi/catkin_ws/build
Devel Space:        [exists] /home/pi/catkin_ws/devel
Install Space:      [unused] /home/pi/catkin_ws/install
Log Space:          [exists] /home/pi/catkin_ws/logs
Source Space:       [exists] /home/pi/catkin_ws/src
DESTDIR:            [unused] None
-------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
-------------------------------------------------------
Additional CMake Args:       -DCMAKE_BUILD_TYPE=Release
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
-------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
-------------------------------------------------------
Workspace configuration appears valid.
-------------------------------------------------------
pi@antonio:~/catkin_ws $ catkin build
-------------------------------------------------------
Profile:                     default
Extending:          [cached] /opt/ros/melodic
Workspace:                   /home/pi/catkin_ws
-------------------------------------------------------
Build Space:        [exists] /home/pi/catkin_ws/build
Devel Space:        [exists] /home/pi/catkin_ws/devel
Install Space:      [unused] /home/pi/catkin_ws/install
Log Space:          [exists] /home/pi/catkin_ws/logs
Source Space:       [exists] /home/pi/catkin_ws/src
DESTDIR:            [unused] None
-------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
-------------------------------------------------------
Additional CMake Args:       -DCMAKE_BUILD_TYPE=Release
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
-------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
-------------------------------------------------------
Workspace configuration appears valid.
-------------------------------------------------------
[build] Found '1' packages in 0.0 seconds.                                       
[build] Package table is up to date.                                             
Starting  >>> mpu9250_9dof_imu                                                   
Finished  <<< mpu9250_9dof_imu                [ 0.9 seconds ]                    
[build] Summary: All 1 packages succeeded!                                       
[build]   Ignored:   None.                                                       
[build]   Warnings:  None.                                                       
[build]   Abandoned: None.                                                       
[build]   Failed:    None.                                                       
[build] Runtime: 1.0 seconds total.                                              
pi@antonio:~/catkin_ws $

robert-meiner, 2020-06-17:
ps. I have no idea why some of this appears in boldface and some in strikethrough characters.  I'm just copy/pasting from the terminal window

NikolausDemmel, 2020-06-17:
This is [markdown syntax](https://guides.github.com/features/mastering-markdown/). For code or output blocks, you should enclose them in three backticks: ```. I fixed your comment.

Seems like more of a catkin / ros question. Maybe you can go back to ros / catkin / catkin-tools tutorials for a better understanding. The packages are found by looking for `package.xml` files. If your `dso_ros` clone in `src` isn't found, it probably doesn't have a `package.xml`. Are you using my `catkin` branch?

robert-meiner, 2020-06-17:
yes.  I cloned your catkin branch into catkin_ws/src.  Here is the git status:

pi@antonio:~/catkin_ws/src/dso_ros $ git status
On branch master
Your branch is up to date with 'origin/master'.

nothing to commit, working tree clean
pi@antonio:~/catkin_ws/src/dso_ros $ git remote -v
origin	https://github.com/NikolausDemmel/dso_ros.git (fetch)
origin	https://github.com/NikolausDemmel/dso_ros.git (push)
pi@antonio:~/catkin_ws/src/dso_ros $

robert-meiner, 2020-06-17:
ayayay... I had checked the "catkin" box, too, before cloning...

I just re-did this, checking the "catkin" branch box, before cloning.  

I still get this:
```
pi@antonio:~/catkin_ws/src/dso_ros $ git status
On branch master
Your branch is up to date with 'origin/master'.

nothing to commit, working tree clean
pi@antonio:~/catkin_ws/src/dso_ros $ 
```

robert-meiner, 2020-06-17:
Thanks for the tips. Success!!

Somehow that link on the catkin branch page leads to the master branch, not the catkin branch.  I manually changed the CMakeLists.txt file and manually added the packages.xml file.  Then I manually input the path to dso:
``` export DSO_PATH=/catkin_ws/src/dso```
Then used the installation method you recommended with catkin tools.

whew...  Now the actual fun begins!

closed 2020-06-17 by robert-meiner

NikolausDemmel, 2020-06-17:
I recommend a git tutorial. You can `git clone -b catkin ...`. Or after cloning change the branch with `git checkout catkin`.