JakobEngel/dso_ros #52: rosbuild error

Dr-coder-001, 2022-09-27:
Can someone help me how to solve this issue?

[rosbuild] Building package segment
[rosbuild] Error from directory check: /opt/ros/noetic/share/ros/core/rosbuild/bin/check_same_directories.py  /home/amirabb/Desktop/segment
1
Traceback (most recent call last):
  File "/opt/ros/noetic/share/ros/core/rosbuild/bin/check_same_directories.py", line 46, in <module>
    raise Exception
Exception
CMake Error at /opt/ros/noetic/share/ros/core/rosbuild/private.cmake:99 (message):
  [rosbuild] rospack found package "segment" at "", but the current directory
  is "/home/amirabb/Desktop/segment".  You should double-check your
  ROS_PACKAGE_PATH to ensure that packages are found in the correct
  precedence order.
Call Stack (most recent call first):
  /opt/ros/noetic/share/ros/core/rosbuild/public.cmake:177 (_rosbuild_check_package_location)
  CMakeLists.txt:12 (rosbuild_init)

-- Configuring incomplete, errors occurred!
See also "/home/amirabb/Desktop/CMakeFiles/CMakeOutput.log

Dr-coder-001, 2022-09-27:
@NikolausDemmel Hey Nikolaus! can you help me with this error? thanks

NikolausDemmel, 2022-09-28:
Sorry, i have no idea.